sharandac/My-TTGO-Watch #128: Touch I2C and crashing web pages

datacute, 2020-10-01:
I'm playing with a later version of the TTGO_TWatch_Library...
The watch is polling the touch screen in the loop.
The Wire1 interface is used by focaltech.cpp without mutexes.
Attempting to have a web page to monitor the touch parameters, results if very frequent crashes.
By altering the focaltech implementation and TTGO.h to use another I2CBus instance wrapping Wire1, the crashes by frequently refreshing web pages goes away.
My guess is that the touch code is using polling because of lack of concurrency controls, and my hope is that by correcting that, we can get lots of improvements.

sharandac, 2020-10-01:
Oh... i always thought that the touch has its own i2c bus (pin 23 and 32) and only the pmu, bma and rtc share one bus (pin 21 and 22).

datacute, 2020-10-01:
Touch does have its own i2c bus. The web page I was playing with was providing touch details as I was exploring touch power saving options. There's nothing preventing concurrent use of the touch i2c bus, and with constant polling it's really easy to conflict.

closed 2020-10-08 by sharandac