sharandac/My-TTGO-Watch #134: Error during build of My-TTGO-Watch in PlatformIO

cyberchucktx, 2020-10-15:
I just got a 

FileNotFoundError: [Errno 2] No such file or directory: '/Users/clindahl/Produced/Hax/TWATCH/My-TTGO-Watch-master/.pio/build/ttgo-t-watch/.sconsign38.dblite'

When I tried to build the project downloaded from GITHUB. 

First time I've set up PlatformIO (at all). 

Am I doing something obviously wrong here? 

Attachment is full log from the Terminal (console) within PlatformIO. 

Thanks, I just want to get load firmware onto my new TTGO Watch ... thanks. 

[platformio-console.txt](https://github.com/sharandac/My-TTGO-Watch/files/5382011/platformio-console.txt)

picpic020960, 2020-10-15:
Bonjour ,
you can upload the .bin
and then upgrade the firmware via OTA with wifi of the watch

sharandac, 2020-10-15:
Thanks for the hint! It seems there have been some changes in the libraries in PlatformIO.

closed 2020-10-20 by sharandac