sharandac/My-TTGO-Watch #135: Extend Gadgetbridge Support

datacute, 2020-10-23:
Since we're pretending to be an Espruino Bangle.js ... the message formats that we can support are documented here:
https://www.espruino.com/Gadgetbridge
There are a few that are not yet supported that could be added:
1. "find" - we could turn on the screen, show a page that allows clicking to say you've found the watch, vibrate and make noise - depending on various existing config settings (e.g. don't make noise if sound is turned off, don't vibrate if the motor is turned off).
2. "alarm" - we could support configuring the alarm
3. "weather" (I haven't seen my android phone send this to my watch - does it require another phone app?) - we could supply weather details to show.

jakub-vesely, 2020-10-23:
Unfortunately,  Bangle.js format does not fit My-TTGO-Watch functionality fully. It is possibleto set an alarm including days of week in Gadgetbridge but the information about days of week is not transferred by Bangle.js BLE messages. I also did not find a way how to send alarm clock values back to Gadgetbridge via Bangle.js BLE protocol. 

From my point of view, does not make sense a lot to implement alarm setting for Gadgetbridge because it newer will work properly. 

Do anyone have  an idea how many effort would be to implement and add own communication protocol to Gadgetbridge to we be able to transfer what we want?

datacute, 2020-10-24:
Here are some example find messages:
msg: {"t":"find","n":true}
msg: {"t":"find","n":false}

I see what you mean about the lack of which day(s) the alarm is for:
msg: {"t":"alarm","d":[{"h":18,"m":14}]}
msg: {"t":"alarm","d":[{"h":18,"m":14},{"h":6,"m":30}]}

Weather example, with "Weather notification" app installed:
https://f-droid.org/en/packages/ru.gelin.android.weather.notification/
msg: {"t":"weather","temp":293,"hum":28,"txt":"Sky is clear","wind":18,"wdir":68,"loc":"Christchurch"}
This needs a bit of conversion, especially determining which icon to show.
Unfortunately "Sky is clear" is not one of these texts:
https://openweathermap.org/weather-conditions

Neuroplant, 2020-11-18:
Do you think of something like this?
[FindPhone.zip](https://github.com/sharandac/My-TTGO-Watch/files/5562184/FindPhone.zip)
(still not ready for Beta, but shows the concept)

Crsarmv7l, 2021-08-21:
> Here are some example find messages:
> msg: {"t":"find","n":true}
> msg: {"t":"find","n":false}
>
> I see what you mean about the lack of which day(s) the alarm is for:
> msg: [...]

Ahh gadgetbridge weather was exactly what I was trying to do. Even just as words like this:

    80F
Clear sky

I cant find where in the src ble messages are parsed or I would write those to SPIFF then display them under the watch face.

Crsarmv7l, 2021-08-21:
Not exactly what I was looking for, but src/gui/mainbar/setup_tile/bluetooth_settings/bluetooth_message.cpp
Has a check for if a message is a "notify" message. Here:
`if( !strcmp( doc["t"], "notify" ) )  {`

Might be able to modify that and the "body" parsing throughout for weather messages to display as a message on update. Not pretty but it could work.

`if( !strcmp( doc["t"], "notify" ) ) || ( !strcmp( doc["t"], "weather" ) ) {`

Crsarmv7l, 2022-01-24:
@sharandac @d03n3rfr1tz3 

Not sure if you would be interested in this given you have the weather app, but I wanted to create this so I wouldn't have to use wifi. It resolves current weather and generates a weather widget based on BLE weather messages from Gadgetbridge (or in my case Amazfish). Both night and day icons work.

I basically just expanded on some of the tools already in place in bluetooth_messages.cpp (expanded the src_icon search to encompass weather descriptions with test cases for night else return day icons) and slightly modified bluetooth_messages.h to account for longer weather descriptors. It doesn't interfere with normal BLE messages either (with amazfish).

[image]

[image]

[image]

Give it a try and include it in your project if you want to. Its a straight swap with these two files.

One big issue does remain: The messages get grabbed up in the message chain, which displays the most current message first. If you delete the weather messages newest to earliest the widget will end up displaying the earliest weather. To keep it accurate you have to delete earliest to newest. I haven't come up with a way to prevent that aside from splitting it out of the message chain.

I started a pull a bit ago but I am terrible with github. 

[bluetooth_message.zip](https://github.com/sharandac/My-TTGO-Watch/files/7925578/bluetooth_message.zip)

PR #191 (2021-03-04): Ble activity support
Add support for notifying activity (stepcounter) via BLE.

* [x] notify data
* [x] optimize notification frequency (timer / value threshold)
* [x] support request for realtime step counting message from Gadgetbridge

Fixes #135