sharandac/My-TTGO-Watch #142: How to debug ???

FantasyFactory, 2020-11-02:
How to debug ?
I can't believe that a complex software like this was developed without debugging directly on the TTWatch !
So, there is a way to debug ?
I guess that many parts of software can be developed using some other esp32 with "similar" hardware with JTAG, but I have'nt this.
HOW I CAN DEBUG ?

jakub-vesely, 2020-11-02:
I'm afraid that debugging is not supported neither for the original t-watch. So I expect it would be much more complicated on the 2020 version. 

Of course, the debugging would be beneficial but you still can use tracing - to insert logs to the developed code and monitor what happen. Unfortunately, there is not used a complex log framework where would be possible to turn on/off modules or levels of logging so I personally removed all development logs before merge to the original repository.    

Other possibility would be to create platform independent (or ESP32 dependent) unit tests but there is expected some initial effort.  A test framework have to be chosen and a proof of concept have to be written. 
Esp-Idf supports fff framework as a component so maybe would be usable for the PatformIO project as well. it supports mocking and it is quite usable. Although, my-ttgo-watch code is well structured I'm afraid that to include unit tests to already written code will be quite a huge effort, but at least for a new functionality or new apps still could be beneficial.

sharandac, 2020-11-02:
Unbelievable but true, everything was developed without a debugger. Unfortunately there is no way to connect a JTAG. 15 years C and 2 years ESP32 experience.
The last time I used a JTAG was on an AVR, long time ago.

FantasyFactory, 2020-11-03:
Yes, it is possible with a lot of experience ! I was dealing with firmware 30 years ago (Z80, 8051, ancient and much simpler stuff) and when it was my main job I too could develop complex software (in assembly) using little or no debuggers, but I was also young : D unfortunately now I'm old, I have little time to devote and I don't have the experience you have on esp32, because I only use it for small projects, simple things, it's not my main job now (I manage a team of sw developers, and unfortunately I have few opportunities to get my hands dirty 

So, I'm developing the basic interpreter and its integration with lvgl from pc, then I'll bring everything to TTwatch minimizing the need for debugging, using the usual serial send

dbuggz, 2020-11-05:
ESP32 support in Arduino IDE has Core Debug Level options (None, Error, Warn, Info, Debug, Verbose) which enable Serial debug statements to be complied. 
In Platformio I currently get serial output like this:

> ets Jun  8 2016 00:22:57
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
... etc 
> starting t-watch V1, version: 2020101101 core: 1
> Configure watchdog to 30s: 0
[I][callback.cpp:75] callback_init(): init callback_t structure success for: powermgm
[I][callback.cpp:114] callback_register(): register callback_func for powermgm success (0x400eab1c:display)
[I][callback.cpp:75] callback_init(): init callback_t structure success for: powermgm loop
[I][callback.cpp:114] callback_register(): register callback_func for powermgm loop success (0x400ea9b8:display loop)
[I][splashscreen.cpp:67] splash_screen_stage_one(): use default boot logo
[W][SPIFFS.cpp:54] begin(): SPIFFS Already Mounted!
[I][pmu.cpp:43] pmu_setup(): set target voltage to 4.36V

but I'm not sure if I enabled anything to produce it.

Regards
Dave

dbuggz, 2020-11-05:
for Hardware debug, there is 

[ESP debug board](https://docs.platformio.org/en/latest/plus/debug-tools/esp-prog.html)
Dave

FantasyFactory, 2020-11-05:
Thanks. Unfortunately the TTwatch does not have the Jtag port, regarding debug output and stacktrace, they are enabled and practically from what I understand they are the only way.

closed 2020-12-02 by sharandac