sharandac/My-TTGO-Watch #143: How to debug ?

FantasyFactory, 2020-11-02:
How to debug ?
I can't believe that a complex software like this was developed without debugging directly on the TTWatch !
So, there is a way to debug ?
I guess that many parts of software can be developed using some other esp32 with "similar" hardware with JTAG, but I have'nt this.
HOW I CAN DEBUG ?

closed 2020-11-02 by FantasyFactory

SuCicada, 2022-07-19:
Excuse me. I had the same problem.
I found PlatformIO supported debug.  Bug not supported board as ttgo-t-watch.
[https://docs.platformio.org/en/stable/boards/espressif32/ttgo-t-watch.html
](https://docs.platformio.org/en/stable/boards/espressif32/ttgo-t-watch.html
)Did you solve it? I very need your suggestion.
Thank You very much.

SuCicada, 2022-07-19:
[https://github.com/sharandac/My-TTGO-Watch/issues/142](https://github.com/sharandac/My-TTGO-Watch/issues/142)