sharandac/My-TTGO-Watch #144: Is the project still alive?

5tormChild, 2020-11-16:
@sharandac did a very great and respectable work, he created a working firmware instead of LilyGO.
But more than 3 weeks had passed and the project seems to be dead ATM.
My C knowledge is limited, I could help with Free Pascal / Delphi / Lazarus stuff, but maybe LilyGO could provide the needed resources to keep this project going.
Still, it's impressive how far did this project go from what LilyGO "provided"!

NorthernDIY, 2020-11-16:
I don't think the project is dead at all.

What's likely is that autumn has brought with it additional time requirements from work/school/family and the resurgence of the pandemic.

As soon as I am done with my classes for the semester I intend to jump right back into this.

5tormChild, 2020-11-23:
Thanks, @NorthernDIY!

closed 2020-12-02 by sharandac

madjestic, 2021-01-05:
3 weeks of no commits and you declare the project dead?  Kids, you are funny.  Get a life or send pull requests.

anarchking, 2021-01-06:
@madjestic..... GitHub is for DEVELOPMENT of projects and community, A simple Point in the right direction Would have worked. Like telling them that they could look at history or comments.