sharandac/My-TTGO-Watch #162: Image slipped upside-down

slaff, 2020-12-18:
I am having a problem with the image on the watch being upside-down. Is there a setting that I need to change in order to have the right orientation? If this is a known feature, please lead me to the URL where I can read more.

P.S: Thanks a lot to all people involved in this project for the great job!

[image]

sharandac, 2020-12-18:
You can find the display orientation in the display menu.

[image]

slaff, 2020-12-18:
> the display menu.

Ok, that sounds very good but how to access the display settings? Is there a documentation/tutorial somewhere?

slaff, 2020-12-18:
Found it. Had to scroll one screen down and then one screen to the right. Thank you again for your help and work! Awesome stuff!

closed 2020-12-18 by slaff