sharandac/My-TTGO-Watch #164: [FR]

ETE-Design, 2020-12-29:
I'm really not into programming... But will it be possible to make it work with ESP3D - Firmware / ESP3D-WEBUI - GUI, used to control a 3D Printer, so it can send status and and let you change settings direct from the watch?

https://github.com/luc-github/ESP3D-WEBUI
https://github.com/luc-github/ESP3D

drunkscientist, 2021-02-13:
have only recvd my watch today, but this should definitely be possible. im still going through the watch code but if it can pull videos from youtube theres no reason it shouldnt be able to push/pull data from a local source

ETE-Design, 2021-02-13:
@drunkscientist So is it a task for you?

drunkscientist, 2021-02-15:
@ETE-Design i mean, im still trying to figure out how to even change the background, but it is something i plan to look at once i understand how this works. a wiki might be helpful, but i dont know if anybody has started one or not

drunkscientist, 2021-03-02:
so im rather proud of myself, i dont have anything functional/usable yet but i have a logo/button on the main screen. still trying to decide what exactly to have it do. so far for sure it should display current status (printing or not, time left on print), which can go on a widget like weather, beyond that im not super sure, since i cant imagine changing much 'at a glance' vs using the webUI on a phone/tablet. ill probably do something similar to the IRremote app and let the user configure the serial commands they want to send, maybe find a way to trigger esp3d's macros, but to start im trying to get connection and get current status displayed somewhere.

ETE-Design, 2021-03-03:
@drunkscientist Great to hear you have progress... Go with ESP3D 3.0, cause that is the newest version Luc is developing on, so to be up to date 3.0 is the way to go :-(

drunkscientist, 2021-03-18:
so its still very much a work in progress, but [heres where im at](https://github.com/drunkscientist/My-TTGO-Watch/tree/master/src/app/esp_app): everything still needs to be hard-coded: printer ip and port, as well as custom gcode commands, i tried to keep them pretty high in the file, but 'host' 'port' and 'command' should be fairly recognizabe. tapping the gauge updates the needles for bed/ extruder temp. eventually the gauge should update automatically and the command should be editable from the watch, printer response should be scrolled on the display as well. top left button sends whatever custom command/ displays response, top right tries to update the lcd to say 'hello world' to test connections, but you could edit that command to be whatever. as said im still working on it, but i figured some feedback couldnt hurt

ETE-Design, 2021-03-18:
@drunkscientist Wauv, sound's great :-) Right now my printer is dissembled cause need to do a lot of mods :-) How about webcam support, could that also be added, so it can show a sceenshot, cause liveview wouldn't be possible right?

drunkscientist, 2021-03-18:
live view should be possible, but i have no idea how. i have esp3d running on an 8266-01, no octoprint, no rpi. i do have an esp-cam that is going into a rover, my next project was probably to make the watch the controller + viewer for it. might be cut/paste applicable depending on how your hardware is going to be set up

d03n3rfr1tz3, 2022-01-04:
Hey everyone,

I found this feature request a few weeks ago and thought it is a good idea. Therefore, I started my own app that directly communicates with G-Codes via WiFi with my FlashForge Adventurer 3. Currently it is only hacked together in a few hours and obviously might not be the same case as yours, but the basics should be similar. Maybe having a look at my app helps with your progress, especially with the app structure, configuration and maybe even the GUI, when I finished that.

As always, I started with the example app and copied over some parts (like the configuration) from other existing apps (in this case, my own kodi_remote app). From there on, it was just a matter of implementing the TCP communication and searching for an icon. 😉

https://github.com/d03n3rfr1tz3/TTGO.T-Watch.2020/commit/f48cf5ec3a9ef59bc88d2ed937d69183c764c475

ETE-Design, 2022-01-09:
@drunkscientist Are you still working on this project, would love to use it when I get a new watch :-)

ETE-Design, 2022-01-09:
@d03n3rfr1tz3 Seems like you know how to program... I'm willing to test it you is you semeday would like to play with the watch and ESP3D

drunkscientist, 2022-01-10:
@ETE-Design i have not had a chance to look at it since i went back to work full time. it does seem like it would be a good 'portfolio piece' so i will probably come back to it at some point

d03n3rfr1tz3, 2022-01-10:
I finished my version for now. It is still pretty simple and just shows some information, therefore no commands or controlling of the printer, but hopefully it's still useful. Maybe I'll add more in the future. Here is a screenshot of the result: https://github.com/d03n3rfr1tz3/TTGO.T-Watch.2020/blob/master/images/printer3d.png

Additionally I looked through the ESP3D implementation and available commands, especially on the supported firmwares and tried as much compatibility for them as possible. So it should work with most somehow, hopyfully.
https://github.com/luc-github/ESP3D/blob/ecf48fe89de215052c388a29eb28556bb483911b/wiki/Firmware--support.md
https://raw.githubusercontent.com/wiki/luc-github/ESP3D/docs/Commands3.txt

Biggest problem is, that there are nearly no example results documented. Therefore, I had to guess based on the examples I could find. Temperature should work best, followed by the print progress. Printer type/name and firmware version will probably not work on all printers and the printer state (e.g. READY, BUILDING on my printer) is probably not supported by any other printer than mine. If anything is missing, the corresponding label is just left empty, so that the GUI should still look okish.

d03n3rfr1tz3, 2022-01-28:
Just a small update: A few days ago I also added a simple MJPEG stream functionality on a second page of the app, if a camera is configured. Obviously it is so slow, that it is more like a slideshow and it also does not like incomplete frames or unstable connections, but it somehow works.
https://github.com/d03n3rfr1tz3/TTGO.T-Watch.2020/tree/master/src/app/printer3d