sharandac/My-TTGO-Watch #168: Unable to upload anything...

ETE-Design, 2021-01-06:
Have just recieved my T-Watch-2020... Have followed the instructions, but when I try to upload the "master" I get a lot of errors other than `<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name`, so it fail to upload... Then I have tryed with "Experimental" instead and not I don't get so many errors only theese:
`xtensa-esp32-elf-g++: error: src\my-ttgo-watch.ino.cpp: No such file or directory
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
*** [.pio\build\ttgo-t-watch\src\my-ttgo-watch.ino.cpp.o] Error 1`
How to make it upload?

sharandac, 2021-01-06:
The warnings can be ignored and come from the TTGO library. The last error comes from PlatformIO if indexing has not yet been completed. Just wait a little and try again.

The following video might also be interesting:

[https://www.youtube.com/watch?v=7vOaqnmAkcg](https://www.youtube.com/watch?v=7vOaqnmAkcg)

ETE-Design, 2021-01-06:
@sharandac After 5 min. waiting I get this instead:
`.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\libraries\GxEPD\src\../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/Fonts/GFXFF/GFXFF.h:18:69: fatal error: ../../../Adafruit-GFX-Library/Fonts/FreeMonoBold12pt7b.h: No such file or directory
.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\libraries\GxEPD\src\../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/Fonts/GFXFF/GFXFF.h:18:69: fatal error: ../../../Adafruit-GFX-Library/Fonts/FreeMonoBold12pt7b.h: No such file or directory
compilation terminated.
*** [.pio\build\ttgo-t-watch\libc4d\TTGO TWatch Library\libraries\GxEPD\src\GxEPD.cpp.o] Error 1
.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\libraries\GxEPD\src\GxGDEH0154D67\../../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/Fonts/GFXFF/GFXFF.h:13:52: fatal error: ../../../Adafruit-GFX-Library/gfxfont.h: No such file or directory
*** [.pio\build\ttgo-t-watch\libc4d\TTGO TWatch Library\libraries\GxEPD\src\GxFont_GFX.cpp.o] Error 1
*** [.pio\build\ttgo-t-watch\libc4d\TTGO TWatch Library\libraries\GxEPD\src\GxGDEH0154D67\GxGDEH0154D67.cpp.o] Error 1`

sharandac, 2021-01-06:
Unfortunately, I cannot reproduce this error. I have tried it with a freshly installed PlatformIO. No errors. But I can only test it under Linux, there is no Windows.

ETE-Design, 2021-01-06:
@sharandac Okey, hope someone with Windows can help then :-D

datacute, 2021-01-07:
Mine works on windows. Which part of that path reported as missing is missing?

asvavilov, 2021-01-09:
> No such file or directory compilation terminated

This error from "TTGO TWatch Library" or dependencies.
I have example project in PlatformIO with these compilation errors.
I have not errors in Arduino IDE in a similar example project.

ETE-Design, 2021-01-09:
@asvavilov So how to solve it in VSCode + PlatformIO

ETE-Design, 2021-01-10:
Have now tryed Arduino IDE and opened the my-ttgo-watch.ino in SRC Directory... Even had to rename the directory to MY-TTGO-WATCH... Now I get this error instead...
In file included from C:\Users\XXX\Desktop\T-Watch-2020\My-TTGO-Watch-master\my-ttgo-watch\my-ttgo-watch.ino:53:0:
C:\marlin\sketch\app/powermeter/powermeter_app.h:25:26: fatal error: gui/icon.h: No such file or directory
compilation terminated.

asvavilov, 2021-01-12:
I moved the folder from user documents folder to a folder c:\_tmp and this fixed the compilation issue.

ETE-Design, 2021-01-14:
@sharandac Got it uploaded today with success... But the screen is upside down, and it freeze at "Init Powermgm" :-( Another question if I ever success getting it running, what file to setup Wireless Network, Weather and so... There seems to be nothing in the Wiki :-(

sharandac, 2021-01-15:
@ETE-Design Oh, that shouldn't happen. Do you have a log file? The serial monitor should tell you a lot about what went wrong. All config files are set in the clock itself via the GUI, usually it is not necessary to change anything yourself.

ETE-Design, 2021-01-15:
@sharandac Where to find a log file? It uploaded with success :-) How does' the Watch know my wireless network withouy telling it anything, also weather app need API key, where?

robertafps, 2021-01-15:
> @sharandac Got it uploaded today with success- Got the version from today... But the screen is upside down, and it freeze at "Init Powermgm" :-( Another question if I ever success getting it running, what file to setup Wireless Network, Weather and so... There seems to be nothing in the Wiki :-(
I'm also having this issue

sharandac, 2021-01-16:
@ETE-Design Log from Serial Monitor

robertafps, 2021-01-16:
@sharandac I don't know if it'll help but this is what I got from the serial monitor

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
starting t-watch V1, version: 2021011402 core: 1
Configure watchdog to 30s: 0
[E][json_psram_allocator.h:13] allocate(): allocate zero bytes? really? abort
[E][motor.cpp:131] motor_read_config(): update check deserializeJson() failed: EmptyInput
[I][callback.cpp:76] callback_init(): init callback_t structure success for: powermgm loop
[E][json_psram_allocator.h:13] allocate(): allocate zero bytes? really? abort
[E][display.cpp:180] display_read_config(): update check deserializeJson() failed: EmptyInput
[I][callback.cpp:76] callback_init(): init callback_t structure success for: powermgm
[I][splashscreen.cpp:67] splash_screen_stage_one(): use default boot logo
[W][SPIFFS.cpp:54] begin(): SPIFFS Already Mounted!
[E][json_psram_allocator.h:13] allocate(): allocate zero bytes? really? abort
[E][pmu.cpp:291] pmu_read_config(): update check deserializeJson() failed: EmptyInput
[I][pmu.cpp:48] pmu_setup(): set target voltage to 4.2V
[I][callback.cpp:76] callback_init(): init callback_t structure success for: blectl
[I][bma.cpp:67] bma_setup(): stepcounter not valid. reset
[E][json_psram_allocator.h:13] allocate(): allocate zero bytes? really? abort
[E][bma.cpp:255] bma_read_config(): update check deserializeJson() failed: EmptyInput
[E][json_psram_allocator.h:13] allocate(): allocate zero bytes? really? abort
[E][wifictl.cpp:265] wifictl_load_config(): update check deserializeJson() failed: EmptyInput
[I][wifictl.cpp:507] wifictl_Task(): start`

fhartojo, 2021-01-16:
I'm having the same exact issue, too.  (The watch is showing the porcupine(?) and "init powermgm" below it.)  My serial monitor output is the same as @robertafps.

fhartojo, 2021-01-16:
FWIW, I reverted to rev. e206c0585f9b73550ca63ed691591b27c08663f5 "tales from the guru" and that build runs successfully.

ETE-Design, 2021-01-17:
@sharandac Just tryed "Tales From The Guru" - And that also work here, so seems like there is a fault in the newest version? My SSID is hidden, is it possible to add it somehow? Can't find anywhere to add SSID manual...

sharandac, 2021-01-17:
@ETE-Design @fhartojo I think it is a problem in the error handling from json_psram_allocator.h in condition with an empty SPIFF with no config files.

ETE-Design, 2021-01-17:
@sharandac Do you have a solution for that? Also is it possible to add hidden SSID to the Wireless network?

sharandac, 2021-01-17:
@ETE-Design not at the moment, i think it is a feature for the future and need some overhaul in the wifictl

sharandac, 2021-01-17:
@ETE-Design  can you please try the latest version to see if the hidden networks are visible?

ETE-Design, 2021-01-17:
@sharandac Had to make a visible network, and that way I have now updated to newest version... The WiFi now show some - 4 - network without SSID, how to know the right one, and how to tell the name of it, seems like nok possible to give it the correct name, just the code...

sharandac, 2021-01-17:
@ETE-Design Would you have a screenshot of it for me?

closed 2021-01-21 by sharandac