sharandac/My-TTGO-Watch #170: Standy Mode Issue / mqtt fails

alosbh, 2021-01-07:
Hello, 

I implement a quick MQTT example for the watch, but when it goes to standby mode, I can no longer get messages in the topic. It seems like its due to standby shutting off wifi. But even when I mark WLAN config to stay On when standby (the log even shows "go standby blocked"), I cant receive mqtt messages...

Can anyone help me handling the standby to keep mqtt client working full time? Which modules may affect it besides wifictl ?

Thanks in advance

sharandac, 2021-01-07:
Have you tried whether a ping in standby still works? Another question would be which mqtt library you use and how you react to messages. Just show your code.

alosbh, 2021-01-08:
Hello, Dirk! 
Yes, i can ping the device when in standby.  Im using the PubSubClient library.

The app I'm trying to implement (following the same baseline as the others) :

```
#include <WiFi.h>
#include <PubSubClient.h>
const char* mqtt_server = "192.168.0.8";

WiFiClient espClient;
PubSubClient client(espClient);
void jitsupport_app_task( lv_task_t * task );

void callback(char* topic, byte* message, unsigned int length) {
  Serial.print("Message arrived on topic: ");
  Serial.print(topic);
  Serial.print(". Message: ");
  String messageTemp;
  
  for (int i = 0; i < length; i++) {
    Serial.print((char)message[i]);
    messageTemp += (char)message[i];
  }
  Serial.println();

  if (String(topic) == "esp32/output") {

  }
}

void jitsupport_app_main_setup( uint32_t tile_num ) {

  jitsupport_app_main_tile = mainbar_get_tile_obj( tile_num );
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  _jitsupport_app_task = lv_task_create( jitsupport_app_task, 250, LV_TASK_PRIO_MID, NULL );
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client")) {
      Serial.println("connected");
      // Subscribe
      client.subscribe("esp32/output");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(1000);
    }
  }
}

void jitsupport_app_task( lv_task_t * task ) {
    // put your code her
    
    if (!client.connected()) {
    reconnect();
  }
  client.loop();;
}

```

And this is what the log shows from when the Watch is not in standby (You can see the successful MQTT connection) until it goes standby (and then I cannot receive messages anymore)

I appreciate any help !

Thanks a lot!!

```
[I][wifictl.cpp:523] wifictl_Task(): request wifictl on done
Message arrived on topic: esp32/output. Message: MQTT TEST SUCCESFULL
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e434c:0001:display)
[I][display.cpp:114] display_standby(): go standby
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e54c4:0001:pmu)
[I][pmu.cpp:197] pmu_standby(): enable silence wakeup timer, 3min
[I][pmu.cpp:211] pmu_standby(): go standby, enable 3000mV standby voltage
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e3494:0001:bma)
[I][bma.cpp:109] bma_standby(): go standby
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e830c:0001:wifictl)
[W][wifictl.cpp:208] wifictl_powermgm_event_cb(): standby blocked by "enable on standby" option
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e7d48:0001:touch)
[I][touch.cpp:74] touch_powermgm_event_cb(): go standby
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e7504:0001:timesync)
[I][timesync.cpp:82] timesync_powermgm_event_cb(): go standby
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e6254:0001:rtcctl)
[I][rtcctl.cpp:164] rtcctl_powermgm_event_cb(): go standby
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400d5b50:0001:gui)
[I][gui.cpp:97] gui_powermgm_event_cb(): go standby
[I][mainbar.cpp:215] mainbar_jump_to_tilenumber(): jump to tile 0 from tile 0
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e1ee0:0001:blectl)
[I][blectl.cpp:344] blectl_powermgm_event_cb(): go standby
[I][callback.cpp:196] callback_send(): call powermgm cb (0x400e6ba0:0001:sound)
[I][sound.cpp:130] sound_standby(): go standby
[I][powermgm.cpp:138] powermgm_loop(): Free heap: 79288
[I][powermgm.cpp:139] powermgm_loop(): Free PSRAM heap: 3877392
[I][powermgm.cpp:140] powermgm_loop(): uptime: 52
[I][powermgm.cpp:141] powermgm_loop(): go standby blocked

```

sharandac, 2021-01-08:
I see your problem. In standby all lv_task* will be stopped. See /gui/gui.c line 102. The better way for this situation is to use powermgm_register_loop_cb() from /hardware/powermgm.c. So is it possible to exec your loop in standby.

```
#include <WiFi.h>
#include <PubSubClient.h>
#include "hardware/powermgm.h
const char* mqtt_server = "192.168.0.8";

WiFiClient espClient;
PubSubClient client(espClient);
bool jitsupport_powermgm_event_cb( EventBits_t event, void *arg );

void callback(char* topic, byte* message, unsigned int length) {
  Serial.print("Message arrived on topic: ");
  Serial.print(topic);
  Serial.print(". Message: ");
  String messageTemp;
  
  for (int i = 0; i < length; i++) {
    Serial.print((char)message[i]);
    messageTemp += (char)message[i];
  }
  Serial.println();

  if (String(topic) == "esp32/output") {

  }
}

void jitsupport_app_main_setup( uint32_t tile_num ) {

  jitsupport_app_main_tile = mainbar_get_tile_obj( tile_num );
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  powermgm_register_loop_cb( POWERMGM_SILENCE_WAKEUP | POWERMGM_STANDBY | POWERMGM_WAKEUP, jitsupport_powermgm_event_cb, "jitsupport app loop" );
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client")) {
      Serial.println("connected");
      // Subscribe
      client.subscribe("esp32/output");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(1000);
    }
  }
}

bool jitsupport_powermgm_event_cb( EventBits_t event, void *arg ) {
    // put your code her
    
    if (!client.connected()) {
    reconnect();
  }
  client.loop();;
return( true );
}
```

alosbh, 2021-01-08:
It did worked. The problem really was the lv_task.

Thank you so much @sharandac. !!! Closing it. 

Best regards, Andre!

closed 2021-01-08 by alosbh