sharandac/My-TTGO-Watch #174: Add support for watchface

guyou, 2021-01-23:
Would be useful to have support for watchfaces, letting user to change the look'n'feel of the main screen.
Watch faces example: https://getwatchmaker.com/collection/Square
Would be great to have similar possibilities.

guyou, 2021-01-31:
As references:

* One project with faces in mind: https://github.com/AlexGoodyear/agoodWatch
* Another one with multiple faces: https://github.com/wfdudley/T-watch-2020
* A funny face: https://github.com/wizche/ttgo-smb

guyou, 2021-05-01:
I've just discovered a project with a **HUGE** collection of faces: https://twatchbuilder.com/
But the code seems quite closed...

Perhaps a sort of inspiration for a face framework.

sharandac, 2021-05-04:
spiffs content
```
watchface_dial.png
watchface_hour.png
watchface_hour_s.png
watchface_min.png
watchface_min_s.png
watchface_sec.png
watchface_sec_s.png
```
watchface_theme.json
```json
{
  "hour": {
    "enable": true,
    "x_offset": 0,
    "y_offset": 0
  },
  "min": {
    "enable": true,
    "x_offset": 0,
    "y_offset": 0
  },
  "sec": {
    "enable": true,
    "x_offset": 0,
    "y_offset": 0
  },
  "hour_shadow": {
    "enable": true,
    "x_offset": 5,
    "y_offset": 5
  },
  "min_shadow": {
    "enable": true,
    "x_offset": 5,
    "y_offset": 5
  },
  "sec_shadow": {
    "enable": true,
    "x_offset": 5,
    "y_offset": 5
  }
}
```

guyou, 2021-05-07:
Nice work.
I understand that such faces are quite similar.

What's the next step to propose more original faces?
In order to keep the idea of an fully external configuration (with files and not C/C++ extension) I can imagine two options:
- a sort of configuration or domain specific language, where moder can describe the background file to load, the position and size (and font) of text, format strings... in order to display step count, time...
- a fully programmable interface using lua (or something like that)

For the first one, we can found reference around animated SVG: https://developer.mozilla.org/en-US/docs/Web/SVG/Element/animate

sharandac, 2021-05-08:
the current version has a little support of it. like label keyword, font size, color and format strings

[look here](https://github.com/sharandac/My-TTGO-Watch/blob/master/images/watchface/startrek/watchface_theme.json)

guyou, 2021-05-08:
This is really awesome. Many thanks.

guyou, 2021-05-09:
I'm looking for Watchface editors... 
https://v1ack.github.io/watchfaceEditor/ seems promising

guyou, 2021-05-09:
Once again a big thank for this evolution. I was able to play around in order to add a K2000 comlink face. Funny!

[image]

closed 2021-05-09 by guyou

sharandac, 2021-05-10:
Very nice! I made a [repro](https://github.com/sharandac/My-TTGO-Watchfaces) for the watchfaces. Feel free to add it. The app to install it will be coming soon.

guyou, 2021-05-10:
I was thinking about such watchface-store. My feeling is that another solution is to promote one repo per watchface. This way, no potential licence issue, as each watchface is in its own repository, with the licence decided by its author.
To retrieve them? Look at a GitHub topic: https://github.com/topics/my-ttgo-watch-face
I'm quite sure we can build a webpage fetching all these repo and building a global watchface repository.

guyou, 2021-05-10:
Hey, and I created an other one.
[image]

sharandac, 2021-05-11:
Yes, the idea is good. But then we need a definition of what the repro has to look like so that it is easy to use them. The releases do not really work well. The easiest way would be to always have the following files:

```
watchface.tar.gz
watchface_theme_prev_120px.png
watchface_theme_prev_60px.png
watchface_theme_prev.png
```

So it would be enough to set a link to the repro in the watchface_theme_list.json and there you will find these files to use them in the watchface manager I am writing. I download

[https://github.com/sharandac/My-TTGO-Watchfaces/blob/main/watchface_theme_list.json](https://github.com/sharandac/My-TTGO-Watchfaces/blob/main/watchface_theme_list.json)

directly from the watch, which is now possible with the watch.

guyou, 2021-05-11:
I really don't like the idea of hosting the archive **IN** the repository. In my humble opinion, it is something similar to a chicken-and-egg problem.

What about improving the watchface_theme_list.json structure, allowing an optional link to archive? For example :
```JSON
[
{ "name": "gameboy",
  "url": "https://raw.githubusercontent.com/guyou/my-ttgo-watch-face-gameboy/main/",
  "download": "https://github.com/guyou/my-ttgo-watch-face-gameboy/releases/latest/download/watchface.tar.gz"
},{...}
]
```

I will work on a script to collect all repos with the given topic and build this list.

sharandac, 2021-05-11:
Ok, I understand. Sounds logical. I'm just finishing the first version to try it out. After that, we'll rebuild it accordingly.

sharandac, 2021-05-11:
And almost forgotten! Thank you for your constructive help with this project!

guyou, 2021-05-11:
> And almost forgotten! Thank you for your constructive help with this project!

Thank you you for maintaining this project and keeping it really active.