sharandac/My-TTGO-Watch #178: Code formatting in Visual Studio Code

hutch120, 2021-01-30:
Hi @sharandac 

Firstly, awesome work, and thanks for making this OpenSource!

I just got a couple of these watches and I've been hacking this project a bit (https://github.com/hutch120/My-TTGO-Watch) but when I save the changes in VSCode it formats the code differently than you have.

I find auto code formatting really helpful, but clearly this makes it hard to know what I've changed and makes PRs almost impossible to provide. Do you use code formatting or do you have it disabled?

(Just a side note) The watch is pretty unstable with OTB (Out Of the Box) firmware installed - `after <1 hour, pressing the button was hit and miss if it would wake up and turn the screen on` - so I ended up disabling most of the addons except `weather` which has resulted in a much more stable experience, about a day now without a problem.

my-ttg-watch.ino

```
weather_app_setup();
// stopwatch_app_setup();
// alarm_clock_setup();
// crypto_ticker_setup();
// example_app_setup();
// osmand_app_setup();
// IRController_setup();
// fxrates_app_setup();
// powermeter_app_setup();
// FindPhone_setup();
```

Cheers,
Simon

sharandac, 2021-01-30:
Hi @hutch120,

thank you very much!

I have switched off the code formatting under VS :-) I only use autocomplete and tabs.

For me, the watches run relatively stable. I have a total of 4 watches. The update process still causes problems and the motion controller. What I have noticed is that each watch has its own problems. But it's also really hard and sometimes really funny to squeeze so many functions into the small watch. The ESP32 is really not made for this. For this reason I also used a self-built version of the arduino lib with more features and even more to enable. But the biggest problem is actually the really small internal RAM. This is often the reason for crashes, but this has become a challenge. And I think: challenge accepted :).

Pro tip: turn off the autosync on the apps you don't use. and play a bit with the values in the pmu.json file.

Dirk

hutch120, 2021-01-30:
Ah, no worries... I'll disable code formatting for this repo... cheers.

BTW.. My goal with this tech is to create  [Game B](https://www.gameb.wiki/index.php?title=Get_Started) experiences (anti-fragile, scalable, win-win) starting by developing a [protocol layer / generator function](https://www.gameb.wiki/index.php?title=Generator_Functions) over [esp_now](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html)

closed 2021-01-30 by hutch120