sharandac/My-TTGO-Watch #179: Possible memory leak

guyou, 2021-02-04:
Reading BLE code, it appears there is two successive call to createServer:
* https://github.com/sharandac/My-TTGO-Watch/blob/930a1aa85f3ed30beadec889e85573dcab03115d/src/hardware/blectl.cpp#L265
* https://github.com/sharandac/My-TTGO-Watch/blob/930a1aa85f3ed30beadec889e85573dcab03115d/src/hardware/blectl.cpp#L277

Not sure, but I fear it is a memory leak.

datacute, 2021-02-05:
It looks like line 265 should be removed. The following 2 lines setting encryption and security callbacks don't need the server created first, and its confusing (to me as someone who spends all day reading C#) to have the type on line 265 as it looks like its trying to be a different pServer than that in line 61.

sharandac, 2021-02-05:
Almost right. There are two different scopes in terms of logic. The first one will never use. This is only possible in c++ without a warning :-). I think this is still an artifact of development.

guyou, 2021-02-07:
Merged

PR #180 (2021-02-05): Fix memory leak
Fixes #179
closed 2021-02-07 by guyou