sharandac/My-TTGO-Watch #187: BLE advertising as null

guyou, 2021-02-09:
Sometimes, the watch is advertising itself as "null", both observed on GadgetBridge and gatttool.

sharandac, 2021-02-09:
Oh. an effect I have not yet observed. Does this happen in certain situations or sporadically?

guyou, 2021-02-10:
I configured Gadgetbridge. Now, each time I wakeup the watch (via the tilt or via the physical button) few seconds later my phone (e/OS/ based on LineageOS based on Android) proposes a new association with "null".

When I did a test with gatttool, it reported the MAC address twice, one with "null" and one with "Espruino".

I also observed that the link with gadgetbridge is not persistent: Gadgetbridge reports it is trying to reconnect. I don't know if it is a related bug, an other bug or a sort of feature (to save power).

sharandac, 2021-02-10:
Is bluetooth switched on in standby? I use Gadgetbridge with a "normal" Android version 9. so far without problems. I have also tested when bluetooth is switched off in standby, so far without any further problems. It connects relatively quickly after a wakeup. Good question about what the error depends.

guyou, 2021-02-10:
> Is bluetooth switched on in standby? I use Gadgetbridge with a "normal" Android version 9. so far without problems. I have also tested when bluetooth is switched off in standby, so far without any [...]

I'm not sure to do the right config. I *only* clicked on bluetooth icon in order to have it in green, from sliding config panel.
Where am I suposed to select status of bluetooth in standby mode?

sharandac, 2021-02-10:
Take a look into bluetooth setup by doing a long press on the bluetooth icon. always on is what you need.

[image]

guyou, 2021-02-10:
> Take a look into bluetooth setup by doing a long press on the bluetooth icon. always on is what you need.
> 
> [image]

OK, sorry for that.

sharandac, 2021-02-10:
No problem. We learn something every day. I, for example, learn about the github actions. Thanks for that! For example, you can press the power button for 2 seconds for the quickbar. ;-)

closed 2021-02-22 by sharandac