sharandac/My-TTGO-Watch #188: Add voice assistant

guyou, 2021-02-11:
Nowadays, voice driven home assistant are becoming reality. As TTG T-Watch 2020 has mic and speaker, we can make it a terminal component for such new interactions with Voice Assistant systems.

Main idea:
As a user, when I press-release a button on the watch application, my voice message is transmitted to Voice Assistant and, in return, the feedback is played as sound.

Yes, exactly like the comlink in the Knight-Rider / K2000 TV serie of the 90's. :-)

Technically, of course, the watch is not powerful enougth to process speech-to-text and text-to-speech required by such system. It has to communicate with a relay. Communication can be made by WiFi or Bluetooth (don't know): WiFi is great if Voice Assistant is a Home Assitant. While Bluetooth can be good enougth if relay is a smartphone.

As effective solution, I don't know how proprietary solutions like Google Home and Amazon Alexa can be integrated in such project. But on the open-source side, a project like [Mycroft AI](https://mycroft.ai/) can do the job. See [related discussion](https://community.mycroft.ai/t/virtual-devices/5237/13).

NorthernDIY, 2021-02-11:
Neither TTGO T-Watch 2020 v1 or v2 have a microphone.  The MAX98357A is just an audio output device.

guyou, 2021-02-11:
> Neither TTGO T-Watch 2020 v1 or v2 have a microphone. The MAX98357A is just an audio output device.

You're right, I misunderstood a spec...

closed 2021-02-11 by guyou

datacute, 2021-02-11:
v3 does doesn't it?

NorthernDIY, 2021-02-12:
That does appear to be the case, I didn't know there was even a v3 in the works.

datacute, 2021-02-12:
I saw details of v3 here: https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library
It's also currently available for purchase.