sharandac/My-TTGO-Watch #192: >     * How to change the background of the watch?

drunkscientist, 2021-02-14:
>     * How to change the background of the watch?
take a look at /src/gui/gui.cpp, line 48 and 60. https://lvgl.io/tools/imageconverter is your friend
>     * Are there different themes available?
no, i have no themes implemented

_Originally posted by @sharandac in https://github.com/sharandac/My-TTGO-Watch/issues/51#issuecomment-675898174_

so im looking at lines 48 and 60 and... ill upload a screenshot, but i cannot figure out how to change the background...

drunkscientist, 2021-02-14:
[image]

guyou, 2021-02-20:
For what I understand, the current background are coming from https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/src/imgs

To add yours, you will have two options.
The simplest: change the code around lines 127 in gui.cpp to use your own image (after having it converted in C and added in images directory. Simply replace a `LV_IMG_DECLARE( bg );` and `lv_img_set_src( img_bin, &bg );` with your own value.

The other solution is to add a new value into the list of possible background. This will require more work, in gui.cpp and display_settings.cpp.

Finaly, an other cool solution will be to add the hability to use a raw PNG with, for example, a predefined name.

sharandac, 2021-02-21:
Or the easiest way: simply upload a PNG with a resolution of 240x240 pixels via ftp to the Watch and name it bg.png and set it in the display settings page 2.

drunkscientist, 2021-02-24:
> The other solution is to add a new value into the list of possible background. This will require more work, in gui.cpp and display_settings.cpp.

this was what i originally, but ended up going with what @sharandac  suggested. im starting to realize how much of this is just the lvgl stuff that i still havnt look at enough...

closed 2021-02-24 by drunkscientist