sharandac/My-TTGO-Watch #196: Missing dependencies

emilecantin, 2021-02-23:
I just cloned the repo, and I'm trying to run `pio run -t upload`, and it seems like a dependency disappeared.

It's trying to clone "https://github.com/gianbacchio/ESP8266Spiram/", which is referenced from line 40 of platformio.ini, but that repo does not exist.

My platformio install usually works when building ESPHome firmwares; am I missing something?

emilecantin, 2021-02-23:
I finally solved it by changing that line to use `Gianbacchio/ESP8266_Spiram` instead.

There were also issues with the versions for the following:
```
	Gianbacchio/ESP8266_Spiram
	luc-github/ESP32SSDP
	earlephilhower/ESP8266SAM
	nailbuster/ESP8266FtpServer
```

I solved it by removing the version, but I don't think it's a good long-term solution.

sharandac, 2021-02-23:
Thanks for the hint! Gianbacchio/ESP8266_Spiram sound like an typo. But on the other side, i can't reproduce the issue with the other libs under Linux. What kind of system do you use?

sharandac, 2021-02-23:
Can you test the latest experimental branch?

emilecantin, 2021-02-23:
I'm on macOS. I don't have the error message anymore, but it was something about "blah blah version <commit_sha> does not match version 1.1.1". Basically it was taking the latest commit hash as the version number.

emilecantin, 2021-02-23:
I just tested out the experimental branch, and it looks like the first issue (the typo) is fixed. However, the version issue is still present, here's the actual error message:
```
Converting my-ttgo-watch.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing ESP32SSDP @ >=1.1.1
git version 2.29.2
Clonage dans '/Users/emilecantin/Documents/Projets/Perso/My-TTGO-Watch/.pio/libdeps/ttgo-t-watch/_tmp_installing-4vDeDt-package'...
AssertionError: Package version f902c39 doesn't satisfy requirements >=1.1.1:
  File "/Users/emilecantin/Library/Python/2.7/lib/python/site-packages/platformio/builder/main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/emilecantin/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/emilecantin/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
```

The crucial part seems to be `AssertionError: Package version f902c39 doesn't satisfy requirements >=1.1.1:`

sharandac, 2021-02-23:
interesting. and without specifying the version does it work?

emilecantin, 2021-02-23:
@sharandac Yes, it just checks out the latest master branch from these repos

sharandac, 2021-02-23:
my platformio.ini looks like this. 

```
lib_deps = 
	TTGO TWatch Library@=1.4.1
	ESP Async WebServer@>=1.2.0
	AsyncTCP@>=1.1.1
	ArduinoJson@>=6.15.2
	luc-github/ESP32SSDP
	IRremoteESP8266@>=2.7.10
	PubSubClient@>=2.8
    earlephilhower/ESP8266Audio@>=1.8.1
	earlephilhower/ESP8266SAM@>=1.0
	nailbuster/ESP8266FtpServer@>=1.0.1
```

emilecantin, 2021-02-23:
I have to remove a few version numbers for it to work:
```
lib_deps = 
	TTGO TWatch Library@=1.4.1
	ESP Async WebServer@>=1.2.0
	AsyncTCP@>=1.1.1
	ArduinoJson@>=6.15.2
	luc-github/ESP32SSDP
	IRremoteESP8266@>=2.7.10
	PubSubClient@>=2.8
  earlephilhower/ESP8266Audio
	earlephilhower/ESP8266SAM
	nailbuster/ESP8266FtpServer
```

I just get a bunch of `<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name` warnings, but it does build like this.

emilecantin, 2021-02-23:
Fixed indentation:
```
lib_deps = 
  TTGO TWatch Library@=1.4.1
  ESP Async WebServer@>=1.2.0
  AsyncTCP@>=1.1.1
  ArduinoJson@>=6.15.2
  luc-github/ESP32SSDP
  IRremoteESP8266@>=2.7.10
  PubSubClient@>=2.8
  earlephilhower/ESP8266Audio
  earlephilhower/ESP8266SAM
  nailbuster/ESP8266FtpServer
```

sharandac, 2021-02-23:
Thanks a lot! Is merge to the experimental branch

datacute, 2021-02-24:
Confirming: Latest experimental branch commit compiles for me in Windows.

closed 2021-02-24 by sharandac