sharandac/My-TTGO-Watch #197: Inefficient gadgetbridge message buffer

guyou, 2021-02-23:
Current buffer is a raw char* allocated and reallocated each time a char as to be appended.
https://github.com/sharandac/My-TTGO-Watch/blob/948397698f7bc2ce6ef3dee6d22b0534da6e5b26/src/hardware/blectl.cpp#L139
This seems really inefficient as any BLE message from GadgetBridge will be many bytes long (a serialized JSON).

Initial thought was to use std::string or Arduino String as these classes will conserve their capacity between two messages.
But a previous implementation already use such classes.
https://github.com/sharandac/My-TTGO-Watch/commit/46122f4c1a362b922ebdbccb3eda1bb8ed50a0c7#diff-d6ccf24b3ab2d08835c4101f3ecebcd737d72a95d2c2f8c70873850cee62e25a

@sharandac do you remember what was wrong previously?

I'm thinking about using a std::string with an initial capacity of many bytes (enougth to receive the initial time setting message). I can imagine that the wrong side is that in case of a really huge message received once, the memory will remain allocated.
An other solution is to improve current code, but wrapping it in a class in order to manage capacity and allocating/reallocating not one by one, but five by five. Why five? Because a JSON looks like `{"":}`.

Opinion?

guyou, 2021-02-23:
PS: why the BLECharactristic content is not used directly but copied in a C char array?

sharandac, 2021-02-23:
The reason for this implementation is simple: std::string or Arduino String cannot manage PSRAM and the old implementation used a statically allocated RAM area and thus limited the size. This way, however, they could reach messages of any size, depending only on the physical one. But eventually std::string would do the same and also be very inefficient.
 And another reason is probably that I program almost exclusively in C at the operating system level.

sharandac, 2021-02-23:
> PS: why the BLECharactristic content is not used directly but copied in a C char array?

To enable dispatch via callback and ensure that it is preserved even if the callback function overwrites it. Never trust any other function! :-)

guyou, 2021-02-23:
> The reason for this implementation is simple: std::string or Arduino String cannot manage PSRAM and the old implementation used a statically allocated RAM area and thus limited the size. This way, [...]

I will propose an implementation using PSRAM and a chunk based allocator (IMHO much more efficient here).

guyou, 2021-02-23:
> > PS: why the BLECharactristic content is not used directly but copied in a C char array?
> 
> To enable dispatch via callback and ensure that it is preserved even if the callback function overwrites it. Never trust any other function! :-)

The data dispached to callback come from `gadgetbridge_msg`. The `msg` string seems allocated only to buffer the BLECharacteristic content. Of course, we have to carrefully truxt calling context, but here it seems hard to do otherway: they are the caller and we need to access the data. As allocating and copying data is costly, it seems better to avoid such action TWICE here (one for buffer and one for gadgetbridge_msg).

If we have to be more gentle with caller, I suspect we have to quickly retrieve data and release caller by using an asynchronous processing of the message. No?

sharandac, 2021-02-23:
It's hard to do otherwise as long as the data from gadgetbridge is fragmented and as long as there is no way to access the data from BLECharacteristic byte by byte. hence the caching of the message in msg. And I don't think it's really expensive considering the speed of the ESP32.
Another consideration is to save pCharacteristic->getValue().c_str() in a pointer and then process it according to the msg length.

sharandac, 2021-02-23:
```c
    void onWrite( BLECharacteristic *pCharacteristic ) {
        size_t msgLen = pCharacteristic->getValue().length();
        const char* msg = pCharacteristic->getValue().c_str();

        for ( int i = 0 ; i < msgLen ; i++ ) {
            switch( *msg ) {
                case EndofText:         blectl_delete_gadgetbridge_msg();
                                        log_i("attention, new link establish");
                                        blectl_send_event_cb( BLECTL_CONNECT, (void *)"connected" );
                                        break;
                case DataLinkEscape:    blectl_delete_gadgetbridge_msg();
                                        log_i("attention, new message");
                                        break;
                case LineFeed:          log_i("message complete, fire BLTCTL_MSG callback");
                                        if( gadgetbridge_msg[ 0 ] == 'G' && gadgetbridge_msg[ 1 ] == 'B' ) {
                                            log_i("gadgetbridge message identified, cut down to json");
                                            gadgetbridge_msg[ gadgetbridge_msg_size - 1 ] = '\0';
                                            log_i("msg: %s", &gadgetbridge_msg[ 3 ] );
                                            blectl_send_event_cb( BLECTL_MSG, (void *)&gadgetbridge_msg[ 3 ] );
                                        }
                                        else {
                                            log_i("msg: %s", gadgetbridge_msg );
                                            blectl_send_event_cb( BLECTL_MSG, (void *)&gadgetbridge_msg[ 0 ] );
                                        }
                                        break;
                default:                blectl_add_char_to_gadgetbridge_msg( *msg );
            }
            msg++;
        }
    }
```

PR #199 (2021-02-25): Optimize mem management on ble message reception
Fixes #197
closed 2021-02-25 by sharandac