sharandac/My-TTGO-Watch #202: Can't compile using PlatformIO IDE 2.2.1

woutput, 2021-03-03:
Hi and thank you for providing this software.
Unfortunately I can't compile it.

This is the software I use:
- Visual Studio Code 1.53.2
- PlatformIO IDE 2.2.1

This is the output I get:
```
PS C:\Data\Temp\My-TTGO-Watch-master> pio run -t upload
Processing ttgo-t-watch (platform: espressif32; board: ttgo-t-watch; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-watch.html
PLATFORM: Espressif 32 (2.1.0) > TTGO T-Watch
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016+sha.fb19175
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting my-ttgo-watch.ino

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'C:\users\woutp\.platformio\packages\framework-arduinoespressif32\tools\platformio-build.py'
File "C:\users\woutp\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 43, in <module>
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TTGO TWatch Library> 1.4.1
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <ArduinoJson> 6.17.3
|-- <AsyncTCP> 1.1.1
|-- <ArduinoJson> 6.17.3
|-- <ESP32SSDP> 1.1.1
|-- <IRremoteESP8266> 2.7.15
|-- <PubSubClient> 2.8.0
|-- <ESP8266Audio> 1.8.1
|-- <ESP8266SAM> 1.0.0
|   |-- <ESP8266Audio> 1.8.1
|-- <ESP8266FtpServer> 1.0.1
Building in release mode
Compiling .pio\build\ttgo-t-watch\src\app\FindPhone\FindPhone.cpp.o
<command-line>:0:10: warning: missing whitespace after the macro name
In file included from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/LilyGoWatch.h:286:0,
                 from src/config.h:47,
                 from src\app\FindPhone\FindPhone.cpp:25:
.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/TTGO.h:16:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

compilation terminated.
Compiling .pio\build\ttgo-t-watch\src\app\FindPhone\FindPhone_main.cpp.o
Compiling .pio\build\ttgo-t-watch\src\app\FindPhone\images\eye_200px.c.o
*** [.pio\build\ttgo-t-watch\src\app\FindPhone\FindPhone.cpp.o] Error 1
Compiling .pio\build\ttgo-t-watch\src\app\FindPhone\images\eye_64px.c.o
<command-line>:0:10: warning: missing whitespace after the macro name
In file included from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/LilyGoWatch.h:286:0,
                 from src/config.h:47,
                 from src\app\FindPhone\FindPhone_main.cpp:25:
.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/TTGO.h:16:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

compilation terminated.
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
In file included from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/src/lv_misc/../lv_conf_internal.h:33:0,
                 from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/src/lv_misc/lv_log.h:16,
                 from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/lvgl.h:26,
                 from src\app\FindPhone\images\eye_64px.c:1:
.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lv_conf.h:104:31: fatal error: esp32-hal-psram.h: No such file or directory

*************************************************************************
* Looking for esp32-hal-psram.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp32-hal-psram.h"
* Web  > https://platformio.org/lib/search?query=header:esp32-hal-psram.h
*
*************************************************************************

compilation terminated.
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
In file included from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/src/lv_misc/../lv_conf_internal.h:33:0,
                 from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/src/lv_misc/lv_log.h:16,
                 from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/lvgl.h:26,
                 from src\app\FindPhone\images\eye_200px.c:1:
.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lv_conf.h:104:31: fatal error: esp32-hal-psram.h: No such file or directory

*************************************************************************
* Looking for esp32-hal-psram.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp32-hal-psram.h"
* Web  > https://platformio.org/lib/search?query=header:esp32-hal-psram.h
*
*************************************************************************

compilation terminated.
*** [.pio\build\ttgo-t-watch\src\app\FindPhone\FindPhone_main.cpp.o] Error 1
*** [.pio\build\ttgo-t-watch\src\app\FindPhone\images\eye_64px.c.o] Error 1
*** [.pio\build\ttgo-t-watch\src\app\FindPhone\images\eye_200px.c.o] Error 1
================================================================================================ [FAILED] Took 6.63 seconds ================================================================================================
```

My first guess is that some dependencies are missing from platformio.ini
Could you help me please?

Thank you in advance

robouden, 2021-03-03:
My platformIO.ini:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:ttgo-t-watch]
platform = espressif32
board = ttgo-t-watch
framework = arduino
lib_archive = true
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.embed_txtfiles = 
	src/gui/mainbar/setup_tile/time_settings/timezones.json
monitor_speed = 115200
monitor_filters = 
	default
	esp32_exception_decoder
build_flags = 
	-DCORE_DEBUG_LEVEL=3
	-mfix-esp32-psram-cache-issue
src_filter = 
	+<*>
lib_deps = 
	TTGO TWatch Library@=1.4.1
	ESP Async WebServer@>=1.2.0
	AsyncTCP@>=1.1.1
	ArduinoJson@>=6.15.2
	gianbacchio/ESP8266Spiram@^1.0
	luc-github/ESP32SSDP@>=1.1.1
	IRremoteESP8266@>=2.7.10
	PubSubClient@>=2.8
	https://codeload.github.com/earlephilhower/ESP8266Audio/legacy.tar.gz/22b52e0ed5aa86a5e5704c5c86d435c8e3e233a0
	earlephilhower/ESP8266SAM@^1.0
	nailbuster/ESP8266FtpServer@^1.0.1
```

woutput, 2021-03-03:
Thanks Rob for trying to help me out, unfortunately I still get similar results:

```
PS C:\Data\Temp\sharandac-My-TTGO-Watch-master> pio run -t upload
Processing ttgo-t-watch (platform: espressif32; board: ttgo-t-watch; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-watch.html
PLATFORM: Espressif 32 (2.1.0) > TTGO T-Watch
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016+sha.fb19175
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting my-ttgo-watch.ino

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'C:\users\woutp\.platformio\packages\framework-arduinoespressif32\tools\platformio-build.py'
File "C:\users\woutp\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 43, in <module>    
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TTGO TWatch Library> 1.4.1  
|-- <ESP Async WebServer> 1.2.3  
|   |-- <AsyncTCP> 1.1.1
|   |-- <ArduinoJson> 6.17.3     
|-- <AsyncTCP> 1.1.1
|-- <ArduinoJson> 6.17.3
|-- <ESP8266Spiram> 1.0.0        
|-- <ESP32SSDP> 1.1.1
|-- <IRremoteESP8266> 2.7.15     
|-- <PubSubClient> 2.8.0
|-- <ESP8266Audio> 1.1.3
|   |-- <ESP8266Spiram> 1.0.0    
|-- <ESP8266SAM> 1.0.0
|   |-- <ESP8266Audio> 1.1.3     
|   |   |-- <ESP8266Spiram> 1.0.0
|-- <ESP8266FtpServer> 1.0.1     
Building in release mode
Compiling .pio\build\ttgo-t-watch\src\app\FindPhone\FindPhone.cpp.o
<command-line>:0:10: warning: missing whitespace after the macro name
In file included from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/LilyGoWatch.h:286:0,
                 from src/config.h:47,
                 from src\app\FindPhone\FindPhone.cpp:25:
.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/TTGO.h:16:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

compilation terminated.
Compiling .pio\build\ttgo-t-watch\src\app\FindPhone\FindPhone_main.cpp.o
Compiling .pio\build\ttgo-t-watch\src\app\FindPhone\images\eye_200px.c.o
*** [.pio\build\ttgo-t-watch\src\app\FindPhone\FindPhone.cpp.o] Error 1
Compiling .pio\build\ttgo-t-watch\src\app\FindPhone\images\eye_64px.c.o
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:10: warning: missing whitespace after the macro name
In file included from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/src/lv_misc/../lv_conf_internal.h:33:0,
                 from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/src/lv_misc/lv_log.h:16,
                 from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/lvgl/lvgl.h:26,
                 from src\app\FindPhone\images\eye_64px.c:1:
[... 20 lines cut ...]
* CLI  > platformio lib search "header:esp32-hal-psram.h"
* Web  > https://platformio.org/lib/search?query=header:esp32-hal-psram.h
*
*************************************************************************

compilation terminated.
In file included from .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/LilyGoWatch.h:286:0,
                 from src/config.h:47,
                 from src\app\FindPhone\FindPhone_main.cpp:25:
.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src/TTGO.h:16:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

compilation terminated.
*** [.pio\build\ttgo-t-watch\src\app\FindPhone\FindPhone_main.cpp.o] Error 1
*** [.pio\build\ttgo-t-watch\src\app\FindPhone\images\eye_200px.c.o] Error 1
*** [.pio\build\ttgo-t-watch\src\app\FindPhone\images\eye_64px.c.o] Error 1
================================================================================================ [FAILED] Took 6.03 seconds ================================================================================================
```

sharandac, 2021-03-04:
I cannot reproduce the error. It looks like platformio is not installed correctly. What kind of system are you using?

woutput, 2021-03-05:
I run on Windows 10 (Home, 10.0.19041 build 19041)
I removed MS Visual Studio Code and reinstalled the latest: VSCodeUserSetup-x64-1.54.1.exe
When I started VS Code, it still found PlatformIO and my workspace (configuration of projects).
Unfortunately, still the same errors occur using the platformio.ini provided by robouden.

By the way: thank you for helping me. I hope that in the end this will have a positive impact on the project :)

woutput, 2021-03-07:
@sharandac was right, apparently my installation had problems.
Thank you for helping and I'm sorry to have this bug incorrectly associated with this great project.

closed 2021-03-07 by woutput

sharandac, 2021-03-08:
@woutput Thank you for the information. Maybe that could help others!