sharandac/My-TTGO-Watch #216: Missing event for BLE link READY

guyou, 2021-03-25:
In recent changes, I try to refresh some data on GadgetBridge (batt level, stepcounter) when BLE link become ready. This is just one more event to check if the data must be notified or not.

But things don't go in the right direction. I found lots of bugs in my implementation and fixed them. But still not working as expected.

Finally, I suspect the fault is the status of BLE link.

Currently, the code react to the BLECTL_CONNECT. The BLE messages are imediatly emitted. But they seems not received:processed. Looking deeper at the logs, I found that the messages are emitted after the BLECTL_CONNECT, but before the onAuthenticationComplete.

So I imagine the BLE link is only partially established.
Any opinion on this?

If it's true that the BLE link should not be used before onAuthenticationComplete, I think we have to introduce an other state or event, something like BLECTL_READY.

sharandac, 2021-03-26:
I see what you mean. I hadn't noticed that before. I would change line 74
```c
        blectl_set_event( BLECTL_CONNECT );
        blectl_clear_event( BLECTL_DISCONNECT );
        blectl_send_event_cb( BLECTL_CONNECT, (void *) "connected" );
```
to
```c
        blectl_set_event( BLECTL_ACTIVE );
        blectl_clear_event( BLECTL_DISCONNECT );
        blectl_send_event_cb( BLECTL_ACTIVE, (void *) "active" );
```
and add the following to the onAuthenticationComplete function in line 136

```c
        if ( blectl_get_event( BLECTL_ACTIVE ) ) {
            blectl_clear_event( BLECTL_ACTIVE );
            blectl_set_event( BLECTL_CONNECT );
            blectl_send_event_cb( BLECTL_CONNECT, (void *) "connected" );
        }
```

This should work without any changes in other places and a working connection would then be correctly indicated by BLECTL_CONNECT as it should be.

Any other suggestions?

guyou, 2021-03-27:
Elegant small change. But I see two possible drawbacks.

First, I think it could better for later understanding to keep BLECTL_CONNECT associated with the onConnect.
Then, changing BLECTL_CONNECT can have impact on currently functioning code. But perhaps it could be better as the expected meaning of BLECTL_CONNECT is having a link ready.

Technically, I imagined to:
- keep BLECTL_CONNECT as currently
- **add** the bit BLECTL_ACTIVE when auth is done
And perhaps revise all current usages of BLECTL_CONNECT to ensure the effective expectation.

Not sure... nevertheless.

sharandac, 2021-03-28:
I have now added for testing my version ( BLECTL_READY renamed BLECTL_AUTHWAIT ) and changed the onAuthenticationComplete slightly.
I also noticed that the battery level is not always transmitted immediately when a connection is made. Is this related to the BleUpdate?

closed 2021-04-05 by sharandac

guyou, 2021-04-06:
> I have now added for testing my version ( BLECTL_READY renamed BLECTL_AUTHWAIT ) and changed the onAuthenticationComplete slightly.
> I also noticed that the battery level is not always transmitted immediately when a connection is made. Is this related to the BleUpdate?

Yes, I'm still trying to understand what's wrong with the current implementation.

My feeling is that this serial protocol is useful to easily extend features, but it is not error prone as announcement are not acknowledged (I'm not sure if GB received and accepted to message).
but still investigating...