sharandac/My-TTGO-Watch #222: No power after successful flash

theycallmevirgo, 2021-04-01:
Windows: Win10 Enterprise x64 v1909 

vscode x64 v1.55

pio core v5.1.1

smartwatch - https://www.amazon.com/gp/product/B08NK11BWM/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

Replicate: build and flash project with PlatformIO 

modified platformio.ini follows

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env]
; overrides the default arduino-esp32 framework with an custom built arduino-esp32 framework
; the custom arduino-esp32 framework provides better power managment, dynamic frquency scaling and 80Mhz Flash/SPIRAM support
; built from https://github.com/espressif/esp32-arduino-lib-builder
platform_packages = framework-arduinoespressif32 @ https://github.com/sharandac/arduino-esp32-hedge.git

[env:ttgo-t-watch-v1]
upload_port=COM2
platform = espressif32
board = ttgo-t-watch
framework = arduino
lib_archive = true
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.embed_txtfiles = 
	src/gui/mainbar/setup_tile/time_settings/timezones.json
monitor_speed = 115200
monitor_filters = 
	default
	esp32_exception_decoder
build_flags = 
	-DCORE_DEBUG_LEVEL=3
    -D LILYGO_WATCH_2020_V1
	-mfix-esp32-psram-cache-issue
src_filter = 
	+<*>
lib_deps = 
    TTGO TWatch Library@=1.4.1
    ESP Async WebServer@>=1.2.0
    AsyncTCP@>=1.1.1
    ArduinoJson@>=6.15.2
    luc-github/ESP32SSDP
    IRremoteESP8266@>=2.7.10
    PubSubClient@>=2.8
    earlephilhower/ESP8266Audio
    earlephilhower/ESP8266SAM
    nailbuster/ESP8266FtpServer
```

sharandac, 2021-04-05:
What is meant by: "No power after successful flash" ?

theycallmevirgo, 2021-04-08:
It won't turn on, screen blank and unresponsive, side button does nothing.

sharandac, 2021-04-08:
And what comes on the serial interface?

pdelprat, 2021-04-10:
Same issue with latest binary and the previous, here is the Serial logging output with previous version : 

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x69646e49
invalid header: 0x69646e49
invalid header: 0x69646e49
invalid header: 0x69646e49
invalid header: 0x69646e49
invalid header: 0x69646e49
invalid header: 0x69646e49
invalid header: 0x69646e49
`
I use the 2020 V1 version,
Pascal

sharandac, 2021-04-10:
Thanks, it looks like an hardware issue or a problem while flash the spi flash. It is an msg from the bootloader. Can you build and flash an binary without the following line in platformio.ini and test it?

```
platform_packages = framework-arduinoespressif32 @ https://github.com/sharandac/arduino-esp32-hedge.git
```

Without this line the spi flash works with 40Mhz. And by the way: have you tested another project for the t-watch 2020?

pdelprat, 2021-04-10:
Hello,
Many thanks for your quick answer,
I just received this t-watch,
It works before I try to update the firmware,
I never try another firmware,
I will do,
Pascal

theycallmevirgo, 2021-04-10:
Sorry for the late response, I was able to build and write with pio by downloading the "official" watch library containing lilygowatch.h through the pio interface. However 

- monitoring terminal results in the reboot loop
- disconnecting from USB stops the reboot loop, but the toucscreen accepts no input. The graphical oputput updates as expected, but the touch input is unresponsive. 

Complete log dump after rebooting follows

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1108
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
starting t-watch V1, version: 2021040501 core: 1
Configure watchdog to 30s: 0
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm
[I][framebuffer.cpp:158] framebuffer_setup_dma(): custom arduino-esp32 framework detected, single DMA framebuffer enable ( 9600 bytes )
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm loop
[I][splashscreen.cpp:69] splash_screen_stage_one(): use default boot logo
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[E][vfs_api.cpp:135] remove(): /pmu_charge.csv does not exists or is directory
[I][pmu.cpp:84] pmu_setup(): set target voltage to 4.2V
[I][callback.cpp:89] callback_init(): init callback_t structure success for: blectl
[I][callback.cpp:89] callback_init(): init callback_t structure success for: wifictl
[I][callback.cpp:89] callback_init(): init callback_t structure success for: timesync
[E][callback.cpp:203] callback_send(): no callback structure found
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=0, y=0, id=main tile
[E][main_tile.cpp:249] main_tile_update_time(): maintile not initialized
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=1, y=0, id=app tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=2, y=0, id=app tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=3, y=0, id=app tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=0, y=1, id=note tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=1, y=1, id=setup tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=2, y=1, id=setup tile
[I][callback.cpp:89] callback_init(): init callback_t structure success for: rtctl
[I][callback.cpp:89] callback_init(): init callback_t structure success for: bma
[I][callback.cpp:89] callback_init(): init callback_t structure success for: pmu
[I][callback.cpp:89] callback_init(): init callback_t structure success for: sound
[I][callback.cpp:89] callback_init(): init callback_t structure success for: display
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=0, y=4, id=battery setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=0, y=5, id=battery setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=2, y=4, id=display settings
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=2, y=5, id=display settings
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=4, y=4, id=move settings
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=6, y=4, id=wifi setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=6, y=5, id=wifi setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=7, y=4, id=wifi setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=7, y=5, id=wifi setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=9, y=4, id=bluetooth setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=11, y=4, id=bluetooth pairing
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=13, y=4, id=bluetooth call
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=15, y=4, id=bluetooth message
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=17, y=4, id=bluetooth media
[I][time_settings.cpp:92] time_setting_set_region_location(): timezone = Etc/GMT
[I][time_settings.cpp:93] time_setting_set_region_location(): region = Etc
[I][time_settings.cpp:94] time_setting_set_region_location(): location = GMT
[I][time_settings.cpp:132] time_settings_create_regionlist(): selected region entry = 7
[I][time_settings.cpp:173] time_settings_create_locationlist(): selected location entry = 0
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=19, y=4, id=time setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=21, y=4, id=update setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=21, y=5, id=update setup
[I][callback.cpp:89] callback_init(): init callback_t structure success for: http ota
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=23, y=4, id=Utilities setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=25, y=4, id=sound setup
[... 134 lines cut ...]
[I][callback.cpp:46] callback_print():  |  |--id:http updater, event mask:000f
[I][pmu.cpp:328] pmu_loop(): battery state: 99%, connected, discharge, battery ok (0x0563)
[I][main_tile.cpp:268] main_tile_update_time(): renew date: Sat 06.Feb 2021
[I][main_tile.cpp:278] main_tile_update_time(): renew time: 22:21
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40001277  PS      : 0x00060a30  A0      : 0x800f6591  A1      : 0x3ffe1280
A2      : 0x00000000  A3      : 0x3ffebb30  A4      : 0x00000000  A5      : 0x3ffe13c4
A6      : 0x3ffbdad0  A7      : 0x00060220  A8      : 0x800fff15  A9      : 0x3ffe1250
[... cut ...]
      (inlined by) WiFiGenericClass::_eventCallback(void*, system_event_t*, wifi_prov_event_t*) at C:/.platformio/packages/framework-arduinoespressif32@src-745eb9bf7c21582e61c010425ed055df/libraries/WiFi/src/WiFiGeneric.cpp:475
  #3  0x400ff18c:0x3ffe13b0 in _network_event_task(void*) at c:\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:2114
  #4  0x4009516e:0x3ffe13e0 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...`

theycallmevirgo, 2021-04-10:
Sorry for the late response, I was able to build and write with pio by downloading the "official" watch library containing lilygowatch.h through the pio interface. However

monitoring terminal results in the reboot loop
disconnecting from USB stops the reboot loop, but the toucscreen accepts no input. The graphical oputput updates as expected, but the touch input is unresponsive.
Complete log dump after rebooting follows

`Rebooting...
ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1108
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
starting t-watch V1, version: 2021040501 core: 1
Configure watchdog to 30s: 0
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm
[I][framebuffer.cpp:158] framebuffer_setup_dma(): custom arduino-esp32 framework detected, single DMA framebuffer enable ( 9600 bytes )
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm loop
[I][splashscreen.cpp:69] splash_screen_stage_one(): use default boot logo
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[E][vfs_api.cpp:135] remove(): /pmu_charge.csv does not exists or is directory
[I][pmu.cpp:84] pmu_setup(): set target voltage to 4.2V
[I][callback.cpp:89] callback_init(): init callback_t structure success for: blectl
[I][callback.cpp:89] callback_init(): init callback_t structure success for: wifictl
[I][callback.cpp:89] callback_init(): init callback_t structure success for: timesync
[E][callback.cpp:203] callback_send(): no callback structure found
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=0, y=0, id=main tile
[E][main_tile.cpp:249] main_tile_update_time(): maintile not initialized
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=1, y=0, id=app tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=2, y=0, id=app tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=3, y=0, id=app tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=0, y=1, id=note tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=1, y=1, id=setup tile
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=2, y=1, id=setup tile
[I][callback.cpp:89] callback_init(): init callback_t structure success for: rtctl
[I][callback.cpp:89] callback_init(): init callback_t structure success for: bma
[I][callback.cpp:89] callback_init(): init callback_t structure success for: pmu
[I][callback.cpp:89] callback_init(): init callback_t structure success for: sound
[I][callback.cpp:89] callback_init(): init callback_t structure success for: display
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=0, y=4, id=battery setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=0, y=5, id=battery setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=2, y=4, id=display settings
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=2, y=5, id=display settings
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=4, y=4, id=move settings
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=6, y=4, id=wifi setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=6, y=5, id=wifi setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=7, y=4, id=wifi setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=7, y=5, id=wifi setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=9, y=4, id=bluetooth setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=11, y=4, id=bluetooth pairing
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=13, y=4, id=bluetooth call
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=15, y=4, id=bluetooth message
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=17, y=4, id=bluetooth media
[I][time_settings.cpp:92] time_setting_set_region_location(): timezone = Etc/GMT
[I][time_settings.cpp:93] time_setting_set_region_location(): region = Etc
[I][time_settings.cpp:94] time_setting_set_region_location(): location = GMT
[I][time_settings.cpp:132] time_settings_create_regionlist(): selected region entry = 7
[I][time_settings.cpp:173] time_settings_create_locationlist(): selected location entry = 0
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=19, y=4, id=time setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=21, y=4, id=update setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=21, y=5, id=update setup
[I][callback.cpp:89] callback_init(): init callback_t structure success for: http ota
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=23, y=4, id=Utilities setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=25, y=4, id=sound setup
[I][mainbar.cpp:158] mainbar_add_tile(): add tile: x=25, y=5, id=sound setup
[... 133 lines cut ...]
[I][callback.cpp:46] callback_print(): | |--id:http updater, event mask:000f
[I][pmu.cpp:328] pmu_loop(): battery state: 99%, connected, discharge, battery ok (0x0563)
[I][main_tile.cpp:268] main_tile_update_time(): renew date: Sat 06.Feb 2021
[I][main_tile.cpp:278] main_tile_update_time(): renew time: 22:21
Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x40001277 PS : 0x00060a30 A0 : 0x800f6591 A1 : 0x3ffe1280
A2 : 0x00000000 A3 : 0x3ffebb30 A4 : 0x00000000 A5 : 0x3ffe13c4
A6 : 0x3ffbdad0 A7 : 0x00060220 A8 : 0x800fff15 A9 : 0x3ffe1250
A10 : 0x3ffe1290 A11 : 0x3ffeb64a A12 : 0x00001028 A13 : 0x3ffd60b0
[... cut ...]
(inlined by) WiFiGenericClass::_eventCallback(void*, system_event_t*, wifi_prov_event_t*) at C:/.platformio/packages/framework-arduinoespressif32@src-745eb9bf7c21582e61c010425ed055df/libraries/WiFi/src/WiFiGeneric.cpp:475
#3 0x400ff18c:0x3ffe13b0 in _network_event_task(void*) at c:.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:2114
#4 0x4009516e:0x3ffe13e0 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...`

sharandac, 2021-04-10:
I think I have located the problem. The Json Config loader crashes when the file system does not yet contain any files. I am in the process of ...

One question: are these all new watches?

theycallmevirgo, 2021-04-10:
Firstt one ordered Mar 23, second one delivered last Wednesay.

flecoufle, 2021-04-10:
I have (I think) the same problem (first firmware flash) with a new watch (few month from ali but not flash yet), I test the fix soon...

sharandac, 2021-04-10:
Then please check out the latest firmware 2021041001.

flecoufle, 2021-04-10:
(on my side) I haven't the reboot anymore (with commit  e7f1d69) Thanks!!

pdelprat, 2021-04-11:
Just to let you know,
I'm able to compile & flash every t-watch samples and it works,
Also I'm using this CLI command to flash your binary :
C:\Python27\python.exe C:\utilisateurs\pasca\Documents\Arduino\hardware\espressif\esp32\tools\esptool.py --port COM4 --baud 115200 write_flash 0 '.\ttgo-t-watch2020_v1.ino (3).bin'
Perhaps I'm wrong,
Thanks for your work,
Pascal

commit e7f1d695 (2021-04-10): fix #222
closed 2021-04-13 by sharandac