sharandac/My-TTGO-Watch #227: TT Go Watch V3 support

valioiv, 2021-04-12:
What to change in the code in order to build it and run the firmware on V3 watch?

sharandac, 2021-04-12:
I don't have a V3, but I tried to include it once. Please test it once. Don't forget to change the env to v3.

[image]

[image]

valioiv, 2021-04-12:
@sharandac, thanks for the fast reaction! I managed to build it with your new commit. Will flash it on the Watch to test once I receive my V3 device (hopefully on Friday this week). Will update it soon! Thanks again!

PGNetHun, 2021-04-23:
Hi @valioiv , @sharandac ,
I have compiled and tested your actual code with V3 device, and it throws exception because in [TTGO TWatch Library 1.4.2](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/V1.4.2/src) the `TWATCH_TFT_MISO` constant is set to `GPIO_NUM_MAX` instead of 0:
`#define TWATCH_TFT_MISO             (GPIO_NUM_MAX)`
([link](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/42f9939e300527b0d42f2b5f8651f3508e3eebcf/src/board/twatch2020_v3.h#L6))

The error itself (from serial port):
```
E (3059) spi: spicommon_bus_initialize_io(207): miso not valid
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40094b40
file: ".pio/libdeps/ttgo-t-watch-v3/TTGO TWatch Library/src/libraries/TFT_eSPI/TFT_eSPI.cpp" line 1121
func: bool TFT_eSPI::initDMA()
expression: ret

ELF file SHA256: 0000000000000000

Backtrace: 0x40094628:0x3ffd2ff0 0x40094b43:0x3ffd3010 0x4010aebf:0x3ffd3030 0x400f56e5:0x3ffd30a0 0x400f50c5:0x3ffd30c0 0x400f61ba:0x3ffd30e0 0x400f9343:0x3ffd3110 0x40159ab6:0x3ffd3140 0x4009516e:0x3ffd3160
  #0  0x40094628:0x3ffd2ff0 in invoke_abort at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40094b43:0x3ffd3010 in _esp_error_check_failed at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:721
  #2  0x4010aebf:0x3ffd3030 in TFT_eSPI::initDMA() at .pio/libdeps/ttgo-t-watch-v3/TTGO TWatch Library/src/libraries/TFT_eSPI/TFT_eSPI.cpp:8188
  #3  0x400f56e5:0x3ffd30a0 in framebuffer_setup(bool, bool) at src/hardware/framebuffer.cpp:44
  #4  0x400f50c5:0x3ffd30c0 in display_setup() at src/hardware/display.cpp:200
  #5  0x400f61ba:0x3ffd30e0 in hardware_setup() at src/hardware/hardware.cpp:39
  #6  0x400f9343:0x3ffd3110 in setup() at src/my-ttgo-watch.ino:58
  #7  0x40159ab6:0x3ffd3140 in loopTask(void*) at /home/pg/.platformio/packages/framework-arduinoespressif32@src-745eb9bf7c21582e61c010425ed055df/cores/esp32/main.cpp:18
  #8  0x4009516e:0x3ffd3160 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

So, as long this bug is not fixed, you should disable DMA and double buffer in [displayconfig.h](https://github.com/sharandac/My-TTGO-Watch/blob/6a5729713c7d42adf54ab41e13c9fd75fdceab1c/src/hardware/config/displayconfig.h#L46)

[image]

sharandac, 2021-04-23:
as workaround, can you test the following addional build_flags in the ttgo-t-watch-v3 section in platformio.ini

```
build_flags = 
	-DCORE_DEBUG_LEVEL=3
        -D LILYGO_WATCH_2020_V3
        -D TWATCH_TFT_MISO=GPIO_NUM_0
	-mfix-esp32-psram-cache-issue
src_filter = 
```

let's see if this works ... not pretty, but could work. apart from the warnings

titimoby, 2021-04-25:
I'm both new to T-watch and platformIO but I tried both workarounds:

1. platformio.ini modification suggested by @sharandac compiled but did not worked
2. displayconfig.h modifications suggested by @PGNetHun worked, T-Watch is running.

I still don't know how to see serial port to verify I have same messages as @PGNetHun but without modifications or in case 1 the physical result is a T-watch with vibration motor on without interruption.

If I can help with tests, it's the max I can do for now, I need to learn quite a lot still.

PS: thanks for this project by the way 💖

sharandac, 2021-04-25:
Thanks for you info. I have adapt it and disable DMA on V3 per default. Please test it and give me a feedback if will work or not.

PGNetHun, 2021-04-25:
Hi @sharandac ,
Thank you for your quick fix, I can confirm, it works with V3!
The previous solution (to add `-D TWATCH_TFT_MISO=GPIO_NUM_0` build argument) was not working because the `twatch2020_v3.h` file has always overridden the `TWATCH_TFT_MISO` macro with `GPIO_NUM_MAX`. That's why it must be fixed in that repository: https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/issues/126#issue-866458449

@titimoby: to see the logs of ESP32, open the PlatformIO "Serial Monitor" window.
So, if you are new to PlatformIO, you should do the following (see screenshot):
1. After opening project, select the environment to build
2. Run clean task (if you need to build project from scratch, for some build issues cleaning up build folder helps)
3. Run build
4. Upload built firmware
5. Open serial monitor window: this connects to device via serial port with 115k default baudrate (configurable in platformio.ini)
6. Any time you change code, repeat from step 3 (or step 2 if needed)

[image]

PS:
- `PlatformIO: Upload` command also triggers the build (theoratically), so it is enough to click on `Upload` button after code change
- Before every `Upload` command, close the previously opened monitor window, because it locks serial port, and upload would not work

valioiv, 2021-04-30:
@sharandac, @PGNetHun, I can only confirm that TTGo Watch V3 is running fine with the following commit from the master branch (I've flash it a couple of days ago):

_commit 538b819f183c1bf9523cba15d23ecf626a929ae3_
_Author: sharandac <sharandac@snafu.de>_
_Date:   Sun Apr 25 23:54:03 2021 +0200_
_update osm map app_

closed 2021-04-30 by sharandac