sharandac/My-TTGO-Watch #228: Compilation Failed

fliuzzi02, 2021-04-12:
Hi, this project looks fantastic!

I was trying to compile the firmware on VS Code, using PlatformIO Core 5.1.1, Home 3.3.4 on Windows 10, and this error popped up:

> In file included from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/TFT_eSPI.h:577:0,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/U8g2_for_Adafruit_GFX.h:41,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxFont_GFX.h:32,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxEPD.h:21,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxEPD.cpp:13:
.pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/Fonts/GFXFF/GFXFF.h:22:76: fatal error: ../../../Adafruit-GFX-Library/Fonts/FreeMonoBoldOblique12pt7b.h: No such file or directory
compilation terminated.
In file included from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/TFT_eSPI.h:577:0,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/U8g2_for_Adafruit_GFX.h:41,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxFont_GFX.h:32,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxFont_GFX.cpp:23:
.pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/Fonts/GFXFF/GFXFF.h:22:76: fatal error: ../../../Adafruit-GFX-Library/Fonts/FreeMonoBoldOblique12pt7b.h: No such file or directory
compilation terminated.
In file included from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEP015OC1/../../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/TFT_eSPI.h:577:0,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEP015OC1/../../../U8g2_for_Adafruit_GFX/src/U8g2_for_Adafruit_GFX.h:41,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEP015OC1/../GxFont_GFX.h:32,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEP015OC1/../GxEPD.h:21,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEP015OC1/GxGDEP015OC1.h:18,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEP015OC1/GxGDEP015OC1.cpp:14:
.pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEP015OC1/../../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/Fonts/GFXFF/GFXFF.h:14:65: fatal error: ../../../Adafruit-GFX-Library/Fonts/FreeMono12pt7b.h: 
No such file or directory
compilation terminated.
*** [.pio\build\ttgo-t-watch-v1\lib4c6\TTGO TWatch Library\libraries\GxEPD\src\GxEPD.cpp.o] Error 1
In file included from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEH0154D67/../../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/TFT_eSPI.h:577:0,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEH0154D67/../../../U8g2_for_Adafruit_GFX/src/U8g2_for_Adafruit_GFX.h:41,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEH0154D67/../GxFont_GFX.h:32,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEH0154D67/../GxEPD.h:21,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEH0154D67/GxGDEH0154D67.h:18,
                 from .pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEH0154D67/GxGDEH0154D67.cpp:14:
.pio/libdeps/ttgo-t-watch-v1/TTGO TWatch Library/src/libraries/GxEPD/src/GxGDEH0154D67/../../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/Fonts/GFXFF/GFXFF.h:14:65: fatal error: ../../../Adafruit-GFX-Library/Fonts/FreeMono12pt7b.h: No such file or directory
compilation terminated.
*** [.pio\build\ttgo-t-watch-v1\lib4c6\TTGO TWatch Library\libraries\GxEPD\src\GxGDEP015OC1\GxGDEP015OC1.cpp.o] Error 1
*** [.pio\build\ttgo-t-watch-v1\lib4c6\TTGO TWatch Library\libraries\GxEPD\src\GxFont_GFX.cpp.o] Error 1
*** [.pio\build\ttgo-t-watch-v1\lib4c6\TTGO TWatch Library\libraries\GxEPD\src\GxGDEH0154D67\GxGDEH0154D67.cpp.o] Error 1
==================================================================================================== [FAILED] Took 81.67 seconds ====================================================================================================

Environment      Status    Duration
---------------  --------  ------------
ttgo-t-watch-v1  FAILED    00:01:21.669
=============================================================================================== 1 failed, 0 succeeded in 00:01:21.669 =============================================================================================== 
The terminal process "C:\Users\fliuz\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'ttgo-t-watch-v1'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

fliuzzi02, 2021-04-12:
UPDATE: I tried compiling with VS Code and PlatformIO in Ubuntu and it succeded, i looked ar earlier closed Issues and a guy found a similar problem as mine, wich he managed to solve by reinstalling git, wich didn't work for me (On Win 10)

fliuzzi02, 2021-04-16:
UPDATE: I finally found what caused problems while compiling under windows, it was OneDrive synching the folder during the compliation process, disabling onedrive for that folder solved the problem.

P.S. I am now working for an app to use in combination with OpenCPN to gather navigation data

dervomsee, 2021-04-21:
@fliuzzi02 what was the filesystem, where you placed the My-TTGO-Watch directory?

I had the same problem an my Windows 10 PC. On the NTFS partition everything works, on the ReFS partition i get the same error.
Maybe the main reason for the problem is the max. filename length, depending on different filesystems...

fliuzzi02, 2021-04-21:
> @fliuzzi02 what was the filesystem, where you placed the My-TTGO-Watch directory?
>
> I had the same problem an my Windows 10 PC. On the NTFS partition everything works, on the ReFS partition i get the [...]

On NTFS, everything works

ShotokuTech, 2021-04-22:
I went into OneDrive and watched/waited for all the contents of the My-TTGO-Watch to show as synced before I complied and all was well after that. I like OneDrive. But it is a time drag when compiling in Platformio or Arduino.

fliuzzi02, 2021-04-22:
> I went into OneDrive and watched/waited for all the contents of the My-TTGO-Watch to show as synced before I compiled and all was well after that. I like OneDrive. But it is a time drag when compiling in Platformio or Arduino.

I just moved the project to a different drive

closed 2021-05-04 by sharandac

mvdorp, 2021-06-24:
I had similar compiling problems with T-watch examples under PlatformIO.   It turned out to be a problem with max path lengths, as Dervomsee already suggested. Just renamed the project map to something shorter, and it compiled ok.

F99, 2022-06-05:
Win7: same problem:
"Compiling .pio\build\t-watch2020-v3\libd0f\TTGO TWatch Library\libraries\GxEPD\src\GxGDEP015OC1\GxGDEP015OC1.cpp.o
In file included from .pio/libdeps/t-watch2020-v3/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/TFT_eSPI.h:577:0,
                 from .pio/libdeps/t-watch2020-v3/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/U8g2_for_Adafruit_GFX.h:41,
                 from .pio/libdeps/t-watch2020-v3/TTGO TWatch Library/src/libraries/GxEPD/src/GxFont_GFX.h:32,
                 from .pio/libdeps/t-watch2020-v3/TTGO TWatch Library/src/libraries/GxEPD/src/GxEPD.h:21,
                 from .pio/libdeps/t-watch2020-v3/TTGO TWatch Library/src/libraries/GxEPD/src/GxEPD.cpp:13:
.pio/libdeps/t-watch2020-v3/TTGO TWatch Library/src/libraries/GxEPD/src/../../U8g2_for_Adafruit_GFX/src/../../TFT_eSPI/Fonts/GFXFF/GFXFF.h:14:65: fatal error: ../../../Adafruit-GFX-Library/Fonts/FreeMono12pt7b.h: No such file or directory
compilation terminated."

F99, 2022-06-06:
Moved to C:/. Well, yes better than. But still 4 errors. First one:
".pio/libdeps/t-watch2020-v3/ESP8266Audio/src/AudioOutputI2S.cpp:95:41: error: 'i2s_pin_config_t' has no non-static data member named 'mck_io_num'
         .data_in_num = I2S_PIN_NO_CHANGE};"

myspac0, 2022-06-06:
Yes, I commented these lines because not use at first for me....

Now I have this:

Linking .pio\build\t-watch2020-v2\firmware.elf
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\t-watch2020-v2\firmware.elf] Error 1

and impossible to resolve for me....

Sarah-C, 2022-08-27:
Same xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory issue for me. I've mentioned it around a bit. See what happens. Hopefully not a NIMBY war.