sharandac/My-TTGO-Watch #245: Add time to OSMand application

guyou, 2021-04-24:
After a short trip with the OSMand application, I realized a missing feature: time display.

When using the OSMand application, we keep the application always on. So, we lack the display of current time.

sharandac, 2021-04-25:
like c45e4e6 ? a little bit small, but no more space left

dervomsee, 2021-04-25:
I had the same problem yesterday, but with other apps.
Would it not be better to display the time centered in the status bar?
This would solve the problem for asks all apps.

sharandac, 2021-04-25:
But only for the apps that do not hide the status bar. Maybe it is an option to show the time in the quickbar. It can be activated and deactivated everywhere without leaving the app.

closed 2021-04-30 by sharandac