sharandac/My-TTGO-Watch #248: OTA Update fails

dervomsee, 2021-04-27:
Hi,
the OTA update on my V2 watch always fails.

With your modified platform_package i get the following backtrace:

```
[GZUpdater] Starting update
abort() was called at PC 0x4026986b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40094140:0x3fffe920 0x400943bd:0x3fffe940 0x4026986b:0x3fffe960 0x402698b2:0x3fffe980 0x40268f87:0x3fffe9a0 0x402692c2:0x3fffe9c0 0x40269025:0x3fffe9e0 0x400fc89e:0x3fffea00 0x400fcc9e:0x3fffea40 0x400fefc7:0x3fffea90 0x400ff471:0x3fffeb60 0x400e8596:0x3fffeb80 0x40094c86:0x3fffebe0
  #0  0x40094140:0x3fffe920 in invoke_abort at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x400943bd:0x3fffe940 in abort at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4026986b:0x3fffe960 in __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #3  0x402698b2:0x3fffe980 in std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #4  0x40268f87:0x3fffe9a0 in __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
  #5  0x402692c2:0x3fffe9c0 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
  #6  0x40269025:0x3fffe9e0 in operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32
  #7  0x400fc89e:0x3fffea00 in GzUnpacker::gzUncompress(bool, bool, bool, bool) at src/utils/ESP32-targz/ESP32-targz-lib.cpp:1735
  #8  0x400fcc9e:0x3fffea40 in GzUnpacker::gzStreamUpdater(Stream*, unsigned int, int, bool) at src/utils/ESP32-targz/ESP32-targz-lib.cpp:1735
  #9  0x400fefc7:0x3fffea90 in http_ota_start_compressed(char const*, char const*, int) at src/utils/http_ota/http_ota.cpp:98
  #10 0x400ff471:0x3fffeb60 in http_ota_start(char const*, char const*, int) at src/utils/http_ota/http_ota.cpp:58
  #11 0x400e8596:0x3fffeb80 in update_Task(void*) at src/gui/mainbar/setup_tile/update/update.cpp:319
  #12 0x40094c86:0x3fffebe0 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46
```

When changing to the default platform_package (commenting line 15) i get the error message 
```
[W][ESP32-targz-lib.cpp:1338] gzStreamUpdater(): Insufficient heap to decompress (available:31720, needed:36864), aborting
[E][http_ota.cpp:99] http_ota_start_compressed(): gzStreamUpdater failed with return code #-103
```

Do you already have your V2 model to test this?

dervomsee, 2021-04-27:
Using the upstream ESP32-targz lib fixes it for me --> #249

sharandac, 2021-04-27:
Oh, thank you for pointing that out. I had indeed had problems with the older versions and therefore used a modified version. I'll take over the PR right away.

sharandac, 2021-04-27:
@dervomsee I hope that the V2 will arrive in the next few days and that I will be able to test real GPS :-)

sharandac, 2021-04-27:
@dervomsee Ok, I see why I had taken my own version. The upstream version crashes in the same way, a new operator is to blame. It does not work properly with the arduino-ESP32. I would take back this change and look at it again carefully in the days to come.

```
  #4  0x4026e123:0x3fff8480 in __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
  #5  0x4026e23a:0x3fff84a0 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
  #6  0x4026e1f1:0x3fff84c0 in operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32
  #7  0x40155f3a:0x3fff84e0 in GzUnpacker::gzUncompress(bool, bool, bool, bool) at .pio/libdeps/ttgo-t-watch-v1/ESP32-targz/src/ESP32-targz-lib.cpp:1869
```

sharandac, 2021-04-27:
@dervomsee  I came back to why I set up my own framework. It is a strange memory problem and a poor/limited new constructor implementation. Do you have time for a Jitsi meeting tonight?

dervomsee, 2021-04-29:
Maybe you closed it a bit early. The update still fails in 80%.
With the fifths or sixth try it went to 100% but rebooted itself.
I will try it tomorrow again with a PC connected.

sharandac, 2021-04-29:
Oh ... let's open that then. With what message?

dervomsee, 2021-05-01:
Hmm, i reflashed it several times and now it workes. Maybe my local git was not up to date..

closed 2021-05-01 by dervomsee