sharandac/My-TTGO-Watch #252: Firmware 2021042602 bricked my 2020 V1 watch

lmagyar, 2021-04-28:
After uploading 2021042602, I get constant `A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header`

Steps:
- I was able to build locally and upload with USB cable the 2021042602 version (f07acaf). T-Watch had previously the firmware it was delivered with.
- With this 2021042602 version the USB communication is dead. Upload is impossible, terminal displays garbage.
- I was able to update from http://www.neo-guerillaz.de/ttgo-t-watch2020_v1.version.json, but that is exactly the same firmware I has built locally, so nothing changed.

I've tried with all the USB ports and cables I have, reinstalled the USB driver, nothing changed. I can't decide whether it is a firmware or hardware issue.

But strange, that I can't turn on the Watch after the unsuccessful upload attempt. I have to remove the USB cable to let it turn on by pressing the button on the Watch. It looks like the USB chip can't restart/reset the Watch, it remains constantly turned off until I remove the USB cable.

UPLOAD log
```
PS C:\Users\xxx\Documents\PlatformIO\Projects\My-TTGO-Watch> pio run -t nobuild -t upload --upload-port COM100
Processing ttgo-t-watch-v1 (platform: espressif32; board: ttgo-t-watch; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM100
Uploading .pio\build\ttgo-t-watch-v1\firmware.bin
esptool.py v3.0
Serial port COM100
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
=============================================================== [FAILED] Took 34.47 seconds ===============================================================

Environment      Status    Duration
---------------  --------  ------------
ttgo-t-watch-v1  FAILED    00:00:34.472
========================================================== 1 failed, 0 succeeded in 00:00:34.472 ==========================================================
```

TERMINAL log   !!!!!!!!!   scroll down   !!!!!!!!!   the garbage is around `standby blocked` messages  !!!!!!!!!
```
PS C:\Users\xxx\Documents\PlatformIO\Projects\My-TTGO-Watch> pio device monitor --port COM100 --baud 115200
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Miniterm on COM100  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIOck div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1108
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[I][my-ttgo-watch.ino:56] setup(): starting t-watch V1, version: 2021042602 core: 1
[I][my-ttgo-watch.ino:57] setup(): Configure watchdog to 30s: 0
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /motor.json
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /display.json
[I][framebuffer.cpp:161] framebuffer_setup_dma(): custom arduino-esp32 framework detected, double DMA framebuffer enable ( 2 x 9600 bytes )
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm loop
[I][splashscreen.cpp:69] splash_screen_stage_one(): use default boot logo
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /pmu.json
[E][vfs_api.cpp:135] remove(): /pmu_charge.csv does not exists or is directory
[... 90 lines cut ...]
[I][gpsctl.cpp:217] gpsctl_powermgm_event_cb(): go wakeup
[I][blectl.cpp:327] blectl_powermgm_event_cb(): go wakeup
[I][powermgm.cpp:138] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:149] powermgm_loop(): Free heap: 122836
[I][powermgm.cpp:150] powermgm_loop(): Free PSRAM heap: 3696740
[I][powermgm.cpp:151] powermgm_loop(): uptime: 20
[I][display.cpp:128] display_standby(): go standby
[I][bma.cpp:192] bma_standby(): go standby
[I][wifictl.cpp:485] wifictl_standby(): request wifictl standby
[I][wifictl.cpp:462] wifictl_off(): request wifictl off
[I][wifictl.cpp:468] wifictl_off(): wifictl not active, prevent first run crash
�f�~�^wifictl.cpp:490] wifictl_standby(): request wifictl standby done
[I][touch.cpp:118] touch_powermgm_event_cb(): go standby
[I][timesync.cpp:103] timesync_powermgm_event_cb(): go standby
[I][rtcctl.cpp:150] rtcctl_powermgm_e�fxnt���F�:枀��Ff��ff�X[���m�ե���␆␆�138�f�xf�}��ݕɵ��}␕�~ņ�t_cb(): g��␀␚␆�X���f5
[I][sound.cpp:123] sound_powermgm_event_cb(): go standby
[I][gpsctl.cpp:211] gpsctl_powermgm_event_cb(): go standby
[W][blectl.cpp:320] blectl_powermgm_event_cb(): standby blocked by "enable_on_standby" option
[I][powermgm.cpp:179] powermgm_loop(): Free heap: 122884
[I][powermgm.cpp:180] powermgm_loop(): Free PSRAM heap: 3696740
[I][powermgm.cpp:181] powermgm_loop(): uptime: 21
[I][powermgm.cpp:201] powermgm_loop(): go standby blocked
f��f~␀␆=��␆~��␀␘��`␘�␞␆�/�~��`␆�@�f�~␆�`␆�␆�ffx␀f ␘~␀��f�>`��␀���␆xff�����-
--- exit ---
```

guyou, 2021-04-28:
Not sure if it is the same bug, but when my watch goes in such state, unable to naturally react to USB upload, I found the following bypass:
1. expect the upload process start
2. press the power button, keep it press but release until the end of the USB discovery process (1 or 2 seconds?)

Then, my watch successfully react to the upload process, letting me upload a fixed firmware.

lmagyar, 2021-04-28:
I think my situation is different. I've tried your power button trick at least 30 times, without success.

The USB communication turns off the Watch, but can't turn it on back, and somehow blocks the power button, so I can't turn it on also. :( And certainly I've tried turning off the Watch before the upload, no success, starting the "6 second button press to turn it on" before the upload starts, Connecting........ starts waiting, but after the 6 seconds elapses, the Watch can't turn on. Something blocks it.

Additionally, I can't turn on the Watch not just during the Connecting........_____....._____....._____....._____....._____....._____....._____ phase, but even after that!!! I have to wait approx. 10-15 seconds to be able to turn on the Watch! Or removing the cable after the unsuccessful upload, the Watch restarts immediately. Strange.

Certainly I've tried removing the battery, just to do a "real" power off, nothing changed.

I'm new to the ESP environment, reading the source code around `standby blocked` messages, the firmware changes (or try to change) the clock speed, maybe this causes the garbage on the serial, if the serial timing is related to the CPU clock (I don't know, I'm just guessing, I lack the knowledge). And maybe the faulty serial communication blocks the "boot" process somehow. I just guessing. :(

sharandac, 2021-04-29:
I think this is a hardware problem. The firmware has no way to prevent flashing via serial. The serial port can put the ESP32 into boot mode via its control lines (RTS/DTS). Then its own software runs, which the firmware has no influence on. This can happen at any time and is triggered by the flash tool and is independent of the firmware currently running. It almost looks like this is exactly what is not working properly.

And you are right, the firmware is constantly changing its speed to save power. This then creates unreadable characters on the serial port. But this is completely uninteresting in boot mode for flashing, which is triggered by the serial interface.

lmagyar, 2021-04-29:
Thank you for your quick response!

First I was able to upload/flash via USB, from the original/old firmware (delivered with the watch) to the 2021042602 version, and the USB upload died only after this, that's why I think maybe this is not purely HW related, and can be repaired.

Is it somehow possible, that the CPU clock or serial speed (or any other low level setting) after reset and during boot/flashing is affected by the firmware, or fuses, or anything that happened during the first and only USB upload???

I'm really sad, that I followed the instructions, uploaded the newest firmware, and now my watch is useless, I can't upload anything to it. Uploading custom firmware was the only reason to buy my first smart-watch. :(

sharandac, 2021-04-29:
This is a strange error. Have you ever tried to run the Watch completely without battery, only via USB? This is the setup when I develop software to save battery life time. Maybe you can isolate the problem this way.

The bootloader uses its own CPU speed/clock settings. 80Mhz.

And maybe you should try a skatch from the Arduino IDE with different upload speeds.

lmagyar, 2021-05-06:
Closing the issue, because it is not firmware related.

I've spent days debugging this issue, the workaround is to add a 10uF capacitance over the built-in reset-button inside the T-Watch. As I understand, this is a long known issue with ESP32 boards if they don't have enough capacitance over the EN/Reset line after the dual NPN transistor logic. So this is a hardware/design issue.

I've uninstalled and deleted each and every USB driver I could find on my Windows, reinstalled, nothing changed. My guess is that when the upload with USB worked once, that was a rare timing accident, not the normal behavior.

Adding permanently a capacitance over EN/Reset button or adding a button over GPIO0 is impossible due to the small space inside the Watch (I've held it in position with my hand). So I've tried OTA, but that also stopped working due to some real firmware bug, that see each successful download as an error.

closed 2021-05-06 by lmagyar