sharandac/My-TTGO-Watch #257: No sound is being played on Watch V3

valioiv, 2021-05-04:
I have enabled an alarm but I don't hear any sound. Looks like the software is trying to play audio but nothing is hearable:

```
[I][sound.cpp:250] sound_play_progmem_wav(): playing audio (size 12318) from PROGMEM 
[I][sound.cpp:150] sound_powermgm_loop_cb(): stop playing wav sound
[I][sound.cpp:250] sound_play_progmem_wav(): playing audio (size 12318) from PROGMEM 
[I][sound.cpp:150] sound_powermgm_loop_cb(): stop playing wav sound
[I][sound.cpp:250] sound_play_progmem_wav(): playing audio (size 12318) from PROGMEM 
[I][sound.cpp:150] sound_powermgm_loop_cb(): stop playing wav sound
```

Is it some V3 specific again (may be some power to the audio chip is not enabled properly) or the issue is reproducible on V1 watch too (don't have V1 to test)?
I'm using the latest commit at that moment: 503dc4a000df1dd560242bbae0a13866257f2d3e

sharandac, 2021-05-04:
@valioiv Thanks for your feedback! it seems like the v3 has another hardware wiring. please test the last experimental commit. 503dc4a

valioiv, 2021-05-04:
@sharandac, I've just test with whatever you've suggested - now I can hear the alarm! Will use those commit from the experimental until you mainline them in the master. Thanks for fast response!

commit 036a2154 (2021-05-04): fix #257 and some watchface updates
closed 2021-05-04 by sharandac