sharandac/My-TTGO-Watch #261: Error: The program size (4804929 bytes) is greater than maximum allowed (1310720 bytes)

fhartojo, 2021-05-11:
I've not been able to get a successful build since support for v2 and v3 were added.  All builds have failed with the error message listed in the title.  The full build log is attached.  I'm building it in VSCode.  Any idea what I'm doing wrong?

Thanks.

[My-TTGO-Watch.log](https://github.com/sharandac/My-TTGO-Watch/files/6460523/My-TTGO-Watch.log)

sharandac, 2021-05-11:
Strange. I use VSCode under Unix with Platformio. It sound like an Issue from PlatformIO and the wrong destination platform.

```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  23.6% (used 77488 bytes from 327680 bytes)
Flash: [=======   ]  73.3% (used 4805705 bytes from 6553600 bytes)
```

Here is an great [tutorial](https://www.youtube.com/watch?v=wUGADCnerCs) step by step. I hope it is an install Issue.

fhartojo, 2021-05-12:
Project inspect exits with the same error (i.e., program size too big).

I've not gotten a chance to view the tutorial video, but it's weird that my VSCode setup that's been working suddenly stopped (working).  I wonder if I'm suddenly not using the right compiler?

Thanks.

Torcado, 2021-05-18:
I did have the same issue (using PlatformIO on Linux without VSCode) and I just figured out that it does work if I do not use the custom arduino-esp32 framework. I deleted the custom framework in ~/.platformio/packages so it will be downloaded again – now it works flawlessly. It seems like PlatformIO did miss an update of the custom framework.
Hope it helps.

fhartojo, 2021-05-18:
@Torcado It works!  Thanks a lot, mate!

Yes, I deleted (well, renamed) the ~/.platformio/packages and the build is now successful.

Thanks again.

closed 2021-05-18 by fhartojo