sharandac/My-TTGO-Watch #264: Add LCD font

guyou, 2021-05-15:
In order to design some watchface, it would be useful to have an Liquid Crystal font. Idealy both for digits and characters and in different sizes.

guyou, 2021-05-15:
Looking at https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library they embed some fonts and some of them came from
http://oleddisplay.squix.ch/
This service propose fonts like 7 segments or 14 segments.

sharandac, 2021-05-17:
add "font" option in the label section, possible name is "Ubuntu" or "LCD". Is no font set, Ubuntu is set as default. Font size option is the same. And it is backward compatible.

[image]

guyou, 2021-05-18:
Wonderful! Thanks!

sharandac, 2021-05-19:
add optional "hide_interval" option for labels. a positive hide interval means hide/show interval and a negative hide interval means show/hide interval. so it is possible to make a second pointer or a switching information label.