sharandac/My-TTGO-Watch #270: Cyrillic font doesn't seem to work when enabled

IlyaMZP, 2021-05-25:
When receiving a message or playing back media with cyrillic symbols, only question marks get displayed.
`USE_EXTENDED_CHARSET CHARSET_CYRILLIC` is defined in the config.h.
I'm using the Gadgetbridge version 0.57.0 on android 11 with latest firmware version on the watch
[image]
Is there any known reasons why cyrillic might not work?

sharandac, 2021-05-25:
@IlyaMZP Thank you for your feedback. Unfortunately, Gadgetbridge does not support Cyrillic/unicode when transmitting notifications via bluettooth.

closed 2021-05-25 by IlyaMZP

IlyaMZP, 2021-05-26:
For some reason in the recent versions of Gadgetbridge the transliteration setting is hidden. I had to enable it in the source code and build the Gadgetbridge apk. Now it finally automatically transliterates cyrillic to english characters.

sharandac, 2021-05-26:
Thank you for your research! Is it possible that you build a finished apk and make it available to us here?

IlyaMZP, 2021-05-26:
> Thank you for your research! Is it possible that you build a finished apk and make it available to us here?

[Here is the apk I built](https://mzp.icu/files/gadgetbridge_0.57.0.apk). Of course it's not signed by the original developer, so you'll have to uninstall the original app before installing this apk. Otherwise it fails to install. I didn't make any changes besides enabling the transliteration, but I can't prove it, so it's recommended to build it yourself. I'll try to build new versions and update the link as they come out.