sharandac/My-TTGO-Watch #274: Display upside-down and Wifi-problem

IoTAdri, 2021-06-01:
Hi,
I cloned the git, build and uploaded the firmware to my V3 but I have 2 issues:
- The display is upside down (I can send a picture but it is rather self explaining)
- I am not able to select a Wifi-network: in th drop-down menu I see the symbol, tap to make it green... it says scanning.. after a while it says done.. but never asks for my password and the wifisymbol on the mainscreen stays dim..

please advise what to do..
THNX

sharandac, 2021-06-01:
Thank you for the feedback. A detailed explanation of how everything works can be found here:

[usage](https://github.com/sharandac/My-TTGO-Watch/blob/master/USAGE.md)

IoTAdri, 2021-06-01:
THNX very much for your FAST reply

closed 2021-06-01 by sharandac