sharandac/My-TTGO-Watch #276: After Start-up Screen changed every 5 seconds from start screen to a black screen

BerndBerts, 2021-06-14:
the Black screen  has "---" in the middle and underneath a "00:00"
what doing this Screen.

Thx Bernd

sharandac, 2021-06-15:
the description sounds a bit confusing. which hardware version of the watch is used? which firmware version? could you take a picture/video of it.

BerndBerts, 2021-06-15:
hello,
hardware version is V2 with GPS. firmware version is actual the 2021060801
[image]
[image]

https://user-images.githubusercontent.com/85887687/122016912-c86e8300-cdc1-11eb-9dbf-f4af14750e05.mp4

sharandac, 2021-06-15:
Thank you for the feedback. Unfortunately I can't see which app it is in the first screenshot, I suspect the alarm clock is going crazy. Have you looked at the log files on the serial interface? Maybe you can see there which app it is.

BerndBerts, 2021-06-15:
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1108
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[I][my-ttgo-watch.ino:54] setup(): starting t-watch V2, version: 2021060801 core: 1
[I][my-ttgo-watch.ino:55] setup(): Configure watchdog to 30s: 0
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /motor.json
[I][motor.cpp:95] motor_setup(): Motor init: I2C device found at address 0x5a
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /display.json
[I][framebuffer.cpp:165] framebuffer_setup_dma(): custom arduino-esp32 framework detected, single DMA framebuffer enable ( 4800 bytes )
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm loop
[I][splashscreen.cpp:69] splash_screen_stage_one(): use default boot logo
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /pmu.json
[E][vfs_api.cpp:135] remove(): /pmu_charge.csv does not exists or is directory
[I][pmu.cpp:84] pmu_setup(): set target voltage to 4.2V
[I][callback.cpp:89] callback_init(): init callback_t structure success for: blectl
[E][callback.cpp:203] callback_send(): no callback structure found
[I][bma.cpp:65] bma_setup(): stepcounter not valid. reset
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /bma.json
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /wificfg.json
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /timesync.json
[I][callback.cpp:89] callback_init(): init callback_t structure success for: wifictl
[I][callback.cpp:89] callback_init(): init callback_t structure success for: timesync
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /rtcctr.json
[E][callback.cpp:203] callback_send(): no callback structure found
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /blectl.json
[I][callback.cpp:89] callback_init(): init callback_t structure success for: pmu
[I][callback.cpp:89] callback_init(): init callback_t structure success for: rtctl
[E][main_tile.cpp:260] main_tile_update_time(): maintile not initialized
[I][callback.cpp:89] callback_init(): init callback_t structure success for: bma
[I][callback.cpp:89] callback_init(): init callback_t structure success for: pmu
[I][callback.cpp:89] callback_init(): init callback_t structure success for: display
[I][widget_factory.cpp:233] wf_get_settings_header_title(): header_title found
[I][callback.cpp:89] callback_init(): init callback_t structure success for: http ota
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /watchface/watchface_theme.json
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /watchface/watchface_theme.json
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /watchface.json
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /osmmap.json
[I][osmmap_app_main.cpp:696] osmmap_add_tile_server_list(): set osmmap url: OSM Standard, http://a.tile.openstreetmap.org/$z/$x/$y.png
[I][callback.cpp:89] callback_init(): init callback_t structure success for: touch
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /activity.json
[I][calendar_db.cpp:170] calendar_db_exec(): sql: SELECT version FROM calendar_db_version;
[I][calendar_db.cpp:150] calendar_db_version_callback(): calendar_db_version = 1
[I][calendar_db.cpp:180] calendar_db_exec(): query time = 26ms
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /ir-remote.json
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /fx-rates.json
[E][json_psram_allocator.h:17] allocate(): allocate 0 bytes (0x0) json psram failed
[E][json_psram_allocator.h:19] allocate(): allocate zero bytes? really? abort
[E][powermeter_app.cpp:110] powermeter_load_config(): update check deserializeJson() failed: EmptyInput
[I][wifictl.cpp:444] wifictl_on(): request wifictl on
[I][wifictl.cpp:551] wifictl_Task(): start wifictl task, heap: 173748
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /sound.json
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /gpsctl.json
[I][wifictl.cpp:569] wifictl_Task(): request wifictl on done
[I][BLEDevice.cpp:569] getAdvertising(): create advertising
[I][hardware.cpp:89] hardware_post_setup(): Total heap: 228800
[I][hardware.cpp:90] hardware_post_setup(): Free heap: 93888
[I][hardware.cpp:91] hardware_post_setup(): Total PSRAM: 4053916
[I][hardware.cpp:92] hardware_post_setup(): Free PSRAM: 3659388

sharandac, 2021-06-15:
Ok, booting looks normal. And when the flickering is going on, what is on the serial?

BerndBerts, 2021-06-15:
der Log ist aktuell während dieses Zustandes gemacht worden...
..äh sorry the log was actually made during this state.

sharandac, 2021-06-15:
Thank you.Have you ever tried to reset the watch to the factory settings? this should be done by formatting the spiff with the utilities.

BerndBerts, 2021-06-15:
yes i have and reflashed

BerndBerts, 2021-06-15:
ok i have now the alarm clock app tested and set a alarm time.. now the time is on the "black screen" showing and a round clock underneath.. 

the alarm app produce the "error"

sharandac, 2021-06-15:
I suspected as much. But I can't quite figure out why this is happening. I suspect that the RTC is not yet set correctly.

BerndBerts, 2021-06-15:
hmmm.. i have set the actual time in the gui an is showing correct

sharandac, 2021-06-15:
The synchronisation is only happened on standby or when you wake up. I am trying to reproduce the error, so i can fix this issue.

BerndBerts, 2021-06-15:
ok nice thx .. now my log is now showing :

I][rtcctl.cpp:125] set_next_alarm(): next local alarm time: 11:58 day: 22
[I][rtcctl.cpp:127] set_next_alarm(): next GMT0 alarm time: 09:58 day 22
[I][rtcctl.cpp:114] set_next_alarm(): local time: 12:05 day: 15
[I][rtcctl.cpp:125] set_next_alarm(): next local alarm time: 11:58 day: 22
[I][rtcctl.cpp:127] set_next_alarm(): next GMT0 alarm time: 09:58 day 22
[I][rtcctl.cpp:114] set_next_alarm(): local time: 12:05 day: 15
[I][rtcctl.cpp:125] set_next_alarm(): next local alarm time: 11:58 day: 22
[I][rtcctl.cpp:127] set_next_alarm(): next GMT0 alarm time: 09:58 day 22

sharandac, 2021-06-15:
Oh, thanks for the hint. It is more an debug msg and not an error msg. does the error actually still occur after setting the alarm?

BerndBerts, 2021-06-15:
I thought it would also be done by setting the alarm time, but unfortunately nothing changes.
After setting the alarm, instead of the '---' the alarm time is there and vice versa when removing it again '---' see screenshot at the top, otherwise nothing changes.

let me know how I can help you further

sharandac, 2021-06-15:
can you add the following lines of code into src/hardware/rtcctl.cpp @ line 52 into the setup routine? this clears all clocks and timer that generates a interrupt.

```c
    TTGOClass *ttgo = TTGOClass::getWatch();
    ttgo->rtc->clearTimer();
    ttgo->rtc->disableTimer();
    ttgo->rtc->disableCLK();
```

and compile and flash. the last idea is that a timer or clk out is preprogrammed and call the alarm. when it is not work i think it is an hardware problem.

BerndBerts, 2021-06-15:
wow really nice .. phew no hardware problem!
that's it! Standby is now also active.
The solution works !! You can include that as a fix.

big thx
Best regards
Bernd

sharandac, 2021-06-15:
commit on next version

closed 2021-06-15 by sharandac