sharandac/My-TTGO-Watch #280: ANCS BLE notifications for iOS devices

armSeb, 2021-06-23:
Hello,

Many thanks for this great software. While trying to get this device working on iOS, I noticed that there is an arduino library to subscribe and receive to iOS notifications via BLE connection. 

Library on GitHub: https://github.com/Smartphone-Companions/ESP32-ANCS-Notifications

So I plan to create a small app to use it, maybe a simple app who allows to select a BLE device and activate / deactivate the notifications. 

I think that it can be very useful as many people use iOS devices, even if this is not an open source friendly device. 

Do you have some advise or guidelines to create this app ?

drunkscientist, 2021-06-25:
depends on how you want to start, i can see 2 possible first step options: get ancs/ble working with a standard esp32, learn lvgl. you can probably retool another app to get it to do the things, so i suggest you get the basic notification code worked out before you start writing an app. the mistake i made was trying to do both at once, i didnt know if the issue was with lvgl or with my communication

armSeb, 2021-07-06:
I first tried to get the notifications working. To proceed, I added the code directly in the .ino file and it works like a charm.
From iOS I see a device with the name defined in the "begin" method, when I associate to it, iOS asks me if I want to allow this device to receive notifications. Then, each notification is sent to the device.

It seems to be complicated to integrate it as an app because : 

- It requires that the bluetooth is enabled (even when the watch is sleeping).
- The device name and connection security pin is already handled by the ttgo subsystem.

To my mind the library requires some modifications and integration in the native OS code. Also, maybe there is a way to keep the bluetooth LE active without battery drain, to be able to receive notifications even if the screen is off ?