sharandac/My-TTGO-Watch #283: OTA-update only work after disabling weather app sync

twister077, 2021-07-02:
Hi,

Not really an issue if you disable weather app sync, but apparently the update process has some interference from the weather sync process. I tried to update with the weather sync active and the update crashed 4 times in a row. When i disabled the weather app sync it worked fine.

sharandac, 2021-07-02:
Yes, this is a known error. The internal RAM of the ESP is actually too small for so many programs.
This also happens when the ftp server or the web server is active. If these two servers are not needed they should be switched off and reboot so that settings is not loaded.

twister077, 2021-07-02:
Is it possible to auto disable these services during update?

Thx

sharandac, 2021-07-02:
Unfortunately not. Once the servers are started, there is no way to get them out of RAM again, except after a reboot.

twister077, 2021-07-02:
Oke. Maybe updating the usage github page to let other know this behavior. Thx.

closed 2021-07-02 by twister077