sharandac/My-TTGO-Watch #289: i2s.c source code request

valioiv, 2021-07-15:
I'm conducting experiments with the speakers of Watch v3 and got the following kernel panic when using `i2s_write()` in a separate task. I want to see what is happening on line 1330 in `i2s.c` but don't know where to get the exact source code of the drivers like `i2s.c`.
@sharandac , can yo help on that?

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40246bc5  PS      : 0x00060830  A0      : 0x800d5d28  A1      : 0x3ffdbb50  
A2      : 0x00000001  A3      : 0x3ffc5530  A4      : 0x00000500  A5      : 0x3ffdbb8c  
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x800d6983  A9      : 0x3ffdbb30  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00006400  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40246bc5:0x3ffdbb50 0x400d5d25:0x3ffdbb80 0x40094406:0x3ffdbbb0
  #0  0x40246bc5:0x3ffdbb50 in i2s_write at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1330
  #1  0x400d5d25:0x3ffdbb80 in mictest_app_data_to_codec_task(void*) at src/app/mictest_app/mictest_app_main.cpp:864
  #2  0x40094406:0x3ffdbbb0 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

void-spark, 2021-07-15:
ESP IDF source? https://github.com/espressif/esp-idf/blob/master/components/driver/i2s.c

valioiv, 2021-07-15:
> ESP IDF source? https://github.com/espressif/esp-idf/blob/master/components/driver/i2s.c

@void-spark , nope... As you can see there is no line 1330 in the `HEAD` of `esp-idf.` I've checked even older versions of `esp-idf 4.x, 3.x` and there is no matching line 1330 in `i2s_write()`. Probably @sharandac uses some custom tuned sources built by himself and would be great to shared them if possible.

void-spark, 2021-07-15:
Ah, you want the source to https://github.com/sharandac/arduino-esp32-hedge/tree/main/tools/sdk/lib :)
Readme mentions:
# Arduino core with some extra things
- enabled PM/DFS with IDLE Task support
- reduce Wifi TX/RX Buffer for more internal RAM
- enhance flash/psram speed like 80Mhz

Doesn't sound like he changed the i2s files, but who knows :)
My feeling is he used https://github.com/espressif/esp32-arduino-lib-builder and only modified the sdkconfig.

Could it be the line number is after the preprocessor worked or so?

void-spark, 2021-07-15:
Hmm, actually, https://github.com/espressif/esp32-arduino-lib-builder would use release/v3.3 branch around the time that  https://github.com/sharandac/arduino-esp32-hedge was created.
Problem is that release/v3.3 changes all the time, as they merge bug fixes into it.
Around that time would be: https://github.com/espressif/esp-idf/blob/33ad4b6f84d0a232a48f749fcfc2aecc79804d7e/components/driver/i2s.c
On that commit the version is 3.3.4 (it's not the 3.3.4 tag, just what they have in the makefile), which does match ​https://github.com/sharandac/arduino-esp32-hedge/blob/main/tools/sdk/include/esp32/esp_idf_version.h
That version does at least have a line 1330, but it doesn't match the method from the stack trace, so dunno :)

sharandac, 2021-07-18:
The Arduino core was built without any intervention in the sources. Only the build options were adapted. The error also looks more like a wrong use of the libraries. Maybe a look into the sources from you will help. And it would be interesting to know if you have deactivated the sound drivers? Otherwise they will interfere with each other.

closed 2021-08-12 by sharandac