sharandac/My-TTGO-Watch #290: No wifi networks seem to be detected (stuck in scan)

theycallmevirgo, 2021-07-19:
Even my dedicated wifi hotspot cannot be found. Perhaps there is some way to list all available networks?

sharandac, 2021-07-19:
In the wifi settings you can see all visible networks when wifi is enabled. But remember that the watch only sees networks in the 2.4GHz band.

theycallmevirgo, 2021-07-19:
The bandwidth is probably the issue. But, how do I get to that settings page? I've never seen it.

sharandac, 2021-07-19:
see [here](https://github.com/sharandac/My-TTGO-Watch/blob/master/USAGE.md)

theycallmevirgo, 2021-07-19:
OK, if you mean the "main" wlan config accessible from settings tile, that's blank. No networks listed, and I know there's at least one 2.4g in the area.

sharandac, 2021-07-19:
Oh, that's interesting. To be on the safe side: is wifi switched on? Which firmware version is installed?

theycallmevirgo, 2021-07-19:
Version is latest (downloaded yesterday). Wifi is on. Watch is V2 purchased about 16 days ago, built with pio V2 build task.

sharandac, 2021-07-19:
Ok. I've heard from some people that the GPS and WiFi antennas cause problems. There were problems with the antenna connectors. Could you try to go very close to the WiFi router and maybe even test GPS?

sharandac, 2021-07-19:
The two soldering points of the antenna were in connection with the metal housing of the watch

[image]
[image]
[image]

theycallmevirgo, 2021-07-19:
Yeah now you mention it gps doesn't seem to do anything either. Fuck, now I
need to figure out how to take it back to stock to see if they'll rma it.

closed 2021-07-22 by theycallmevirgo