sharandac/My-TTGO-Watch #292: No display after flash

theycallmevirgo, 2021-08-07:
Flashed with latest vscode and pio, serial seems to say it keeps resetting 

Seems like a few people on Telegram have seen this

Representative serial dump follows 

`Serial monitor says; rst:0x10 (RTCWDT_RTC_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-1
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-1
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-1
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-1
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-`

sharandac, 2021-08-07:
Which Watch version is used and which firmware version? Was it compiled under Windows, Mac or Linux?

I miss the first output via the serial of the firmware, so I think that it does not seem to be a problem of the firmware.

Torcado, 2021-08-09:
I am having the same issue:

- T-Watch 2020 v1
- Firmware is current master branch
- PlatformIO 5.1.1 (did also a test with 5.2.0b1)
- Manjaro Linux (Kernel 5.13.5)

Never had this issue before. Already deleted the whole firmware directory and also everything related to PlatformIO. Uploading sample apps via Arduino IDE works.
I also did a checkout of an older firmware version (2 months ago; a build that was already running on my watch) – still a no go … so I guess @sharandac is right in assuming that it is not directly firmware related.

Torcado, 2021-08-09:
Okay, here's the fix …

This issue is related to platform-espressif32 v3.3.1. By using v3.3.0 the problem is eliminated.
Open up your platformio.ini and change `platform = espressif32` to `platform = espressif32@3.3.0`.

commit 37225076 (2021-08-09): fix #292