sharandac/My-TTGO-Watch #293: Rebooting when try to change settings

hideo213, 2021-08-10:
Hy
I have a T-Watch-2020-V3
i have issues with the master Branch firmware (downloaded yesterday)
Complied and uploaded with PlatformIO in VSC on a Linux (deb11) box.

When i try to change settings in the Bluetooth settings menu or in the watchface menu the device reboots imediately.

Is this a known issue, is there a solution?

Regards

Timo

sharandac, 2021-08-10:
Thanks for your feedback! Would you have a complete serial log up to the point when the watch reboots?

hideo213, 2021-08-10:
Hy

I think the interesting part will be the first few lines:

[I][powermgm.cpp:151] powermgm_loop(): uptime: 32
[I][wifictl.cpp:570] wifictl_Task(): request wifictl on done
[I][basejsonconfig.cpp:186] save(): json config serializeJson() success: /pmu.json
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060630  A0      : 0x80101a18  A1      : 0x3ffd7a30  
A2      : 0x00000040  A3      : 0x0000003c  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd79f0  
A10     : 0x3f8897e8  A11     : 0x3f698887  A12     : 0x3ffd7ad0  A13     : 0x00000020  
A14     : 0x00008000  A15     : 0x00008000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000040  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400014fd:0x3ffd7a30 0x40101a15:0x3ffd7a40 0x40101bd2:0x3ffd7a60 0x4010aa03:0x3ffd7ab0 0x400fffb2:0x3ffd7b10 0x400ea469:0x3ffd7b30 0x40166fa5:0x3ffd7b50 0x40167007:0x3ffd7b90 0x40189c6d:0x3ffd7bb0 0x4018eb6a:0x3ffd7bd0 0x4018fd09:0x3ffd7bf0 0x401757d3:0x3ffd7c40 0x401758e1:0x3ffd7c60 0x400e6a00:0x3ffd7c80 0x401015c5:0x3ffd7ca0 0x401056cf:0x3ffd7cc0 0x40105a0b:0x3ffd7ce0 0x40107cc3:0x3ffd7d00 0x401bbbc4:0x3ffd7d20 0x40094406:0x3ffd7d40

Rebooting...
[I][my-ttgo-watch.ino:59] setup(): starting t-watch V3, version: 2021072902 core: 1
[I][my-ttgo-watch.ino:60] setup(): Configure watchdog to 30s: 0
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /motor.json
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /display.json
[I][framebuffer.cpp:103] framebuffer_setup_nodma(): single framebuffer enable ( 4800 bytes )
[I][callback.cpp:89] callback_init(): init callback_t structure success for: powermgm loop
[I][splashscreen.cpp:69] splash_screen_stage_one(): use default boot logo
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /pmu.json
[E][vfs_api.cpp:135] remove(): /pmu_charge.csv does not exists or is directory
[I][pmu.cpp:84] pmu_setup(): set target voltage to 4.2V
[I][callback.cpp:89] callback_init(): init callback_t structure success for: blectl
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /bma.json
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /wificfg.json
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /timesync.json
[I][callback.cpp:89] callback_init(): init callback_t structure success for: wifictl
[I][callback.cpp:89] callback_init(): init callback_t structure success for: timesync
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /rtcctr.json
[I][basejsonconfig.cpp:186] save(): json config serializeJson() success: /rtcctr.json
[E][callback.cpp:203] callback_send(): no callback structure found
[E][json_psram_allocator.h:17] allocate(): allocate 0 bytes (0x0) json psram failed
[E][json_psram_allocator.h:19] allocate(): allocate zero bytes? really? abort
[E][basejsonconfig.cpp:55] load(): json config deserializeJson() failed: EmptyInput, file: /blectl.json
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /blectl.json
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /sound.json
[I][callback.cpp:89] callback_init(): init callback_t structure success for: pmu
[I][callback.cpp:89] callback_init(): init callback_t structure success for: rtctl
[E][main_tile.cpp:260] main_tile_update_time(): maintile not initialized
[I][callback.cpp:89] callback_init(): init callback_t structure success for: bma
[I][callback.cpp:89] callback_init(): init callback_t structure success for: pmu
[I][callback.cpp:89] callback_init(): init callback_t structure success for: display
[I][callback.cpp:89] callback_init(): init callback_t structure success for: sound
[I][widget_factory.cpp:233] wf_get_settings_header_title(): header_title found
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /update.json
[I][callback.cpp:89] callback_init(): init callback_t structure success for: http ota
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /watchface/watchface_theme.json
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /watchface/watchface_theme.json
[I][basejsonconfig.cpp:149] save(): json config serializeJson() success: /watchface/watchface_theme.json
[I][basejsonconfig.cpp:58] load(): json config deserializeJson() success: Ok, file: /watchface.json
[I][basejsonconfig.cpp:186] save(): json config serializeJson() success: /watchface.json
[I][basejsonconfig.cpp:69] load(): reading json failed, call defaults, file: /osmmap.json
[I][osmmap_app_main.cpp:696] osmmap_add_tile_server_list(): set osmmap url: OSM Standard, http://a.tile.openstreetmap.org/$z/$x/$y.png
[I][callback.cpp:89] callback_init(): init callback_t structure success for: touch
[E][json_psram_allocator.h:17] allocate(): allocate 0 bytes (0x0) json psram failed
[E][json_psram_allocator.h:19] allocate(): allocate zero bytes? really? abort
[E][weather.cpp:214] weather_load_config(): update check deserializeJson() failed: EmptyInput
[... cut ...]
[I][rtcctl.cpp:162] rtcctl_powermgm_event_cb(): go standby
[I][gui.cpp:134] gui_powermgm_event_cb(): go standby
[I][sound.cpp:134] sound_powermgm_event_cb(): go standby
[I][gpsctl.cpp:260] gpsctl_powermgm_event_cb(): go standby
[I][blectl.cpp:331] blectl_powermgm_event_cb(): go standby
[I][powermgm.cpp:179] powermgm_loop(): Free heap: 92900
[I][powermgm.cpp:180] powermgm_loop(): Free PSRAM heap: 3646240
[I][powermgm.cpp:181] powermgm_loop(): uptime: 29
[I][powermgm.cpp:184] powermgm_loop(): go standby
[I]

sharandac, 2021-08-11:
Interesting place for a crash. Did you use the serial monitor of PIO? It would be interesting to see in the backtrace which functions were called last.
do you have the same effect when you install the latest firmware via ota?

hideo213, 2021-08-11:
I used the Arduino IDE serial Monitor
I made an upgrade via OTA, that worked, but the Problem stays the same.

sharandac, 2021-08-12:
If it is possible, can you use the PIO monitor? Then you would see the backtraces correctly.
And by the way, I can't reproduce the error. Could you write a step by step guide on how to generate the error?

Torcado, 2021-08-12:
The same happens to me (T-Watch 2020 v1). Here are the steps to reproduce:

- go to bluetooth settings
- try to disable bluetooth

This is the corresponding log:

`[I][bma.cpp:212] bma_wakeup(): go wakeup
[I][wifictl.cpp:504] wifictl_wakeup(): request wifictl wakeup
[I][wifictl.cpp:445] wifictl_on(): request wifictl on
[I][wifictl.cpp:506] wifictl_wakeup(): request wifictl wakeup done
[I][touch.cpp:124] touch_powermgm_event_cb(): go wakeup
[I][timesync.cpp:110] timesync_powermgm_event_cb(): go wakeup
[I][rtcctl.cpp:166] rtcctl_powermgm_event_cb(): go wakeup
[I][gui.cpp:140] gui_powermgm_event_cb(): go wakeup
[I][sound.cpp:137] sound_powermgm_event_cb(): go wakeup
[I][gpsctl.cpp:265] gpsctl_powermgm_event_cb(): go wakeup
[I][blectl.cpp:335] blectl_powermgm_event_cb(): go wakeup
[I][powermgm.cpp:138] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:149] powermgm_loop(): Free heap: 83976
�f�W�␀��~ʆ�֝͞���␃�N��f␀�n␀␘��fx�f�x��f��3␆)I␂␘��6�␆�M	f␀5�heap: 3649880
[I][powermgm.cpp:151] powermgm_loop(): uptime: 63
[I][wifictl.cpp:570] wifictl_Task(): request wifictl on done
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060c30  A0      : 0x80101a34  A1      : 0x3ffd7a60  
A2      : 0x00000040  A3      : 0x0000003c  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd7a20  
A10     : 0x3f8877ec  A11     : 0x3f6989b3  A12     : 0x3ffd7b00  A13     : 0x00000020  
A14     : 0x00008000  A15     : 0x00008000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000040  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400014fd:0x3ffd7a60 0x40101a31:0x3ffd7a70 0x40101bee:0x3ffd7a90 0x4010aa9f:0x3ffd7ae0 0x40100906:0x3ffd7b40 0x400ea495:0x3ffd7b60 0x401671cd:0x3ffd7b80 0x4016722f:0x3ffd7bc0 0x40189e95:0x3ffd7be0 0x4018ed92:0x3ffd7c00 0x4018ff31:0x3ffd7c20 0x401759fb:0x3ffd7c70 0x40175b09:0x3ffd7c90 0x400e6a0c:0x3ffd7cb0 0x401015e1:0x3ffd7cd0 0x401057a7:0x3ffd7cf0 0x40105ae3:0x3ffd7d10 0x40107d5f:0x3ffd7d30 0x401bbdec:0x3ffd7d50 0x4009516e:0x3ffd7d70
  #0  0x400014fd:0x3ffd7a60 in ?? ??:0
  #1  0x40101a31:0x3ffd7a70 in ArduinoJson6183_91::StringAdapter<char const*, false, void>::size() const at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:115
      (inlined by) char const* ArduinoJson6183_91::MemoryPool::saveString<ArduinoJson6183_91::StringAdapter<char*, false, void> >(ArduinoJson6183_91::StringAdapter<char*, false, void> const&) at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Memory/MemoryPool.hpp:73
      (inlined by) bool ArduinoJson6183_91::VariantData::storeString<ArduinoJson6183_91::StringAdapter<char*, false, void> >(ArduinoJson6183_91::StringAdapter<char*, false, void>, ArduinoJson6183_91::MemoryPool*, ArduinoJson6183_91::storage_policies::store_by_copy) at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:353
  #2  0x40101bee:0x3ffd7a90 in bool ArduinoJson6183_91::VariantData::setString<ArduinoJson6183_91::StringAdapter<char*, false, void> >(ArduinoJson6183_91::StringAdapter<char*, false, void>, ArduinoJson6183_91::MemoryPool*) at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:115
      (inlined by) bool ArduinoJson6183_91::variantSetString<ArduinoJson6183_91::StringAdapter<char*, false, void> >(ArduinoJson6183_91::VariantData*, ArduinoJson6183_91::StringAdapter<char*, false, void>, ArduinoJson6183_91::MemoryPool*) at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:57
      (inlined by) ArduinoJson6183_91::enable_if<ArduinoJson6183_91::IsString<char*, void>::value, bool>::type ArduinoJson6183_91::convertToJson<char*>(char* const&, ArduinoJson6183_91::VariantRef) at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:137
      (inlined by) ArduinoJson6183_91::Converter<char*, void>::toJson(char* const&, ArduinoJson6183_91::VariantRef) at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:17
      (inlined by) bool ArduinoJson6183_91::VariantRef::set<char>(char*) const at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:98
      (inlined by) ArduinoJson6183_91::MemberProxy<ArduinoJson6183_91::ElementProxy<ArduinoJson6183_91::MemberProxy<ArduinoJson6183_91::JsonDocument&, char const*> >, char const*>& ArduinoJson6183_91::MemberProxy<ArduinoJson6183_91::ElementProxy<ArduinoJson6183_91::MemberProxy<ArduinoJson6183_91::JsonDocument&, char const*> >, char const*>::operator=<char>(char*) at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:58
      (inlined by) blectl_config_t::onSave(ArduinoJson6183_91::JsonDocument&) at src/hardware/config/blectlconfig.cpp:37
  #3  0x4010aa9f:0x3ffd7ae0 in BaseJsonConfig::save() at src/utils/basejsonconfig.cpp:169 (discriminator 2)
  #4  0x40100906:0x3ffd7b40 in blectl_set_autoon(bool) at .pio/libdeps/ttgo-t-watch-v1/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:281
  #5  0x400ea495:0x3ffd7b60 in bluetooth_enable_onoff_event_handler(_lv_obj_t*, unsigned char) at src/gui/mainbar/setup_tile/bluetooth_settings/bluetooth_settings.cpp:175
[... cut ...]
  #15 0x401057a7:0x3ffd7cf0 in powermgm_send_loop_event_cb(unsigned int) at src/hardware/powermgm.cpp:340
  #16 0x40105ae3:0x3ffd7d10 in powermgm_loop() at src/hardware/powermgm.cpp:340
  #17 0x40107d5f:0x3ffd7d30 in loop() at src/my-ttgo-watch.ino:100
  #18 0x401bbdec:0x3ffd7d50 in loopTask(void*) at /home/hoppe/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:23
  #19 0x4009516e:0x3ffd7d70 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57`

sharandac, 2021-08-12:
Can you test the experimental branch and give me an feedback?

Torcado, 2021-08-12:
Experimental branch does not have this issue.

sharandac, 2021-08-12:
Thanks for your feedback!

hideo213, 2021-08-12:
> Experimental branch does not have this issue.

After some small tests i can confirm that.

Thanks for the Help

sharandac, 2021-08-12:
Update master to version 2021081201.

closed 2021-08-12 by sharandac