sharandac/My-TTGO-Watch #294: Simple Question: BLE Configuration with Gadgetbridge

Crsarmv7l, 2021-08-12:
Hi,

Looking through your code and gadgetbridge, it appears you set up the TTGO to act like/authenticate as if it were a banglejs.

Is that correct?

I am looking to utilize a TTGO with Amazfish, which already supports the banglejs (and would save me some headaches trying to get it working). 

Just wanted to confirm. Thanks

sharandac, 2021-08-12:
Yes, that is correct. I use the bangle.js protocol.

[bangle.js](https://www.espruino.com/Gadgetbridge)

Crsarmv7l, 2021-08-12:
Fantastic thank you!

A few more simple questions:

1. I noticed you have implemented secure bonding/pairing, will it fall back to insecure if necessary?
2. Is the BLE connection stable?

This is looking very promising for my use, especially given all of the features!

closed 2021-08-13 by Crsarmv7l