sharandac/My-TTGO-Watch #295: Time sync & IR Remote

Crsarmv7l, 2021-08-15:
Love the program. I had expected this watch to be much more popular given its an ESP32.

1. The default ir remote doesn't have any remotes and the readme directs us to use wcofigurator. While the github is there, the website to utilize it appears to be down.
2. I would like to also request the implementation of time sync from the companion app vice ntp. It would save a lot of power.

Crsarmv7l, 2021-08-16:
Looking harder at the code the issue with the irremote might be the lack of inclusion of IRremoteESP8266.h and IRsend.h....Dont know if platformio fetched them for me.

It also looks like there should be bluetooth time sync. Any hints on turning off encryption? It might solve my problem since I have to use another application that supports bangle.js but does not (yet) have encryption/authentication

sharandac, 2021-08-17:
> 
>     1. The default ir remote doesn't have any remotes and the readme directs us to use wcofigurator. While the github is there, the website to utilize it appears to be down.

Take a look here -> https://github.com/d03n3rfr1tz3/TTGO.T-Watch.2020/blob/master/conf/ir-remote.json.example by example

> 
>     2. I would like to also request the implementation of time sync from the companion app vice ntp. It would save a lot of power.

per Default time sync work per ntp when wifi is connected and autosync is enabled. Time sync over bluetooth is only active when a device connects via bluetooth/gadgetbridge and time sync is enabled in gadgetbridge.

Crsarmv7l, 2021-08-18:
Yep, I found the bt sync myself after more digging. Thanks for the ir example.

closed 2021-08-18 by Crsarmv7l