sharandac/My-TTGO-Watch #299: Technical help: Bluetooth Messages Implementation issues

Crsarmv7l, 2021-08-24:
Background: Current weather widget relies on http get to provide weather. While a very pretty widget, if I am out weather would not update unless it has wifi on and known accesspoint. Additionally it would provide a big battery drain.

Solution: Gadgetbridge provides a weather message. The simplest implementation is to simply parse these as "notify" messages and take the appropriate part and put it into the main body. I wanted to accomplish this task before moving on to maybe a widget  that simply displays the temp and text description (80 Sky is Clear).

"notify" messages are structured like this: GB({"t":"notify","id":1575479849,"src":"Hangouts","title":"A Name","body":"message contents"})
"weather" messages are structured like this: GB({"t":"weather","temp":"80","hum":"28","txt":"Clear sky","wind":"185",}) there is more but this is the relevant part.

So I decided to simply add recognition of this type of message to the messages app by making simple modifications to src/gui/mainbar/setup_tile/bluetooth_settings/bluetooth_message.cpp

The standard "notify" code is this:
`/*
     * if msg an notify msg?
     */
    if( !strcmp( doc["t"], "notify" ) ) {
        char msg[256];
        serializeJson( doc, msg, sizeof(msg) );
        retval = bluetooth_message_queue_msg( msg );
    }
    return( retval );
}`

Then further down, it addresses how the message displays and what info is parsed here:

`if( !strcmp( doc["t"], "notify" ) ) {
            /*
             * set the receive time string
             */
            struct tm info;
            char timestamp[16]="";
            localtime_r( msg_chain_get_msg_timestamp_entry( bluetooth_msg_chain, entry ), &info );
            int h = info.tm_hour;
            int m = info.tm_min;
            snprintf( timestamp, sizeof( timestamp ), "%02d:%02d", h, m );
            lv_label_set_text( bluetooth_message_time_label, timestamp );
            /*
             * set the numbers of msg string
             */
            snprintf( msg_num, sizeof( msg_num ), "%d/%d", entry + 1, msg_chain_get_entrys( bluetooth_msg_chain ) );
            lv_label_set_text( bluetooth_msg_entrys_label, msg_num );
            lv_obj_align( bluetooth_msg_entrys_label, bluetooth_next_msg_btn, LV_ALIGN_OUT_LEFT_MID, -5, 0 );
            /*
             * hide statusbar
             */
            statusbar_hide( true );
            /*
             * set notify source icon if msg src known
             */
            if ( doc["src"] ) {
                lv_img_set_src( bluetooth_message_img, bluetooth_message_find_img( doc["src"] ) ); 
                lv_label_set_text( bluetooth_message_notify_source_label, doc["src"] );
            }
            else {
                lv_img_set_src( bluetooth_message_img, &message_32px );
                lv_label_set_text( bluetooth_message_notify_source_label, "Message" );
            }
            /*
             * set message if body known or set title and if no other information
             * available set an emty msg
             */
            if ( doc["body"] ) {
                lv_label_set_text( bluetooth_message_msg_label, doc["body"] );
            }
            else if ( doc["title"] ) {
                lv_label_set_text( bluetooth_message_msg_label, doc["title"] );
            }
            else {
                lv_label_set_text( bluetooth_message_msg_label, "" );
            }`

For my supposed to be simple implementation I changed these two sections as follows, which should work but doesnt:
`/*
     * if msg an notify msg?
     */
    if( !strcmp( doc["t"], "notify" ) || !strcmp( doc["t"], "weather") ) {
        char msg[256];
        serializeJson( doc, msg, sizeof(msg) );
        retval = bluetooth_message_queue_msg( msg );
    }
    return( retval );
}`

and

`/*
         * if msg an notify or weather msg?
         */
        if( !strcmp( doc["t"], "notify" ) || !strcmp( doc["t"], "weather" ) ) {
            /*
             * set the receive time string
             */
            struct tm info;
            char timestamp[16]="";
            localtime_r( msg_chain_get_msg_timestamp_entry( bluetooth_msg_chain, entry ), &info );
            int h = info.tm_hour;
            int m = info.tm_min;
            snprintf( timestamp, sizeof( timestamp ), "%02d:%02d", h, m );
            lv_label_set_text( bluetooth_message_time_label, timestamp );` 

For adding weather into the message body with a weather message, I added another else if to the standard message allocations:

`/*
             * set message if body known or set title and if no other information
             * available set an emty msg
             */
            if ( doc["body"] ) {
                lv_label_set_text( bluetooth_message_msg_label, doc["body"] );
            }
            else if ( doc["title"] ) {
                lv_label_set_text( bluetooth_message_msg_label, doc["title"] );
            }
            else if ( doc["temp"]) {
                lv_label_set_text( bluetooth_message_msg_label, doc["temp"] );
            }
            else {
                lv_label_set_text( bluetooth_message_msg_label, "" );
            }`

 I am kind of at a loss as to why this isn't working. I have checked blectl.cpp and didn't see anything. bluejsonrequest.h only has stipulations for conf and request messages.

I would appreciate any help, thank you

sharandac, 2021-08-24:
Thanks for your feedback!

your two changes

```c
if( !strcmp( doc["t"], "notify" ) || !strcmp( doc["t"], "weather") )
```

work correctly so far. at least for me. but there is an error in the output:

```c
lv_label_set_text( bluetooth_message_msg_label, doc["temp"] );
```
remember

```c
doc["temp"]
```

returns in this case an int and not a string. Correct would be

```c
doc["temp"].as<String>().c_str()
```

which returns a string that can be passed to the lv_label_set_text function like this

```c
lv_label_set_text( bluetooth_message_msg_label, doc["temp"].as<String>().c_str() );
```

Or you can do it a bit nicer

```c
            else if ( doc["temp"] ) {
                int temperature = doc["temp"];
                const char temp_str[128] = "";
                snprintf( (char*)temp_str, sizeof( temp_str ), "%s / %d °C / %s", doc["loc"].as<String>().c_str(), temperature - 273, doc["txt"].as<String>().c_str() );
                lv_label_set_text( bluetooth_message_msg_label, temp_str );
            }
```
on the serial interface i see this:

```
[I][blectl.cpp:167] onWrite(): receive 20 bytes msg chunk
[I][blectl.cpp:176] onWrite(): attention, new message
[I][blectl.cpp:167] onWrite(): receive 20 bytes msg chunk
[I][blectl.cpp:167] onWrite(): receive 20 bytes msg chunk
[I][blectl.cpp:167] onWrite(): receive 20 bytes msg chunk
[I][blectl.cpp:167] onWrite(): receive 20 bytes msg chunk
[I][blectl.cpp:167] onWrite(): receive 5 bytes msg chunk
[I][blectl.cpp:179] onWrite(): attention, message complete
[I][blectl.cpp:181] onWrite(): msg: GB({"t":"weather","temp":287,"hum":87,"txt":"Aufgerissene Bew�lkung","wind":0,"wdir":338,"loc":"Jena"})
[I][blectl.cpp:183] onWrite(): gadgetbridge message identified, cut down to json
[I][sound.cpp:386] sound_is_silenced(): no silence sound timeframe
[I][sound.cpp:275] sound_play_progmem_wav(): playing audio (size 12318) from PROGMEM 
[I][bluetooth_message.cpp:591] bluetooth_message_play_audio(): playing default mp3 audio notification
[I][sound.cpp:161] sound_powermgm_loop_cb(): stop playing wav sound
```

and here the result on the screen

[image]

Crsarmv7l, 2021-08-24:
Thank you! That is exactly the sort of implementation I am going for. 

Ultimately I would like to have that sort of thing displayed as a widget under the watch, instead of a message, but that is for another day. I will try your code.

My issue may also be due to utilizing Amazfish as a companion app. But if so I know which is causing my issue.

Crsarmv7l, 2022-01-17:
Took this a step farther. 

Given the txt description, and this https://openweathermap.org/weather-conditions I was able to import the owm images and run doc[txt] through your src_icon find function to find the correct weather image to display with a weather message. Still need to mess with the title alignment, and the image size a bit.

Is it elegant? No. Is it concise? Also no. But it works.
[image]

Ultimately I would like to write weather message information to weather_today. That should allow the weather widget to update either with a bluetooth weather message OR wifi.

Current attempts to write to weather_today haven't been going so well lol.