sharandac/My-TTGO-Watch #300: Powermeter app

fraschizzato, 2021-09-10:
For my purposes the powermeter app could be a good starting point, so I'm trying to use it.
With serial monitor I can't see anything related, the only message is about the json configuration save.
The configuration is correct but I can't see any data that I send in mqtt with pc.
Any idea?

Crsarmv7l, 2021-09-11:
Not quite sure what you are looking for specifically, but you should look at:

https://github.com/d03n3rfr1tz3/TTGO.T-Watch.2020

It is a fork of this repo, if you look in src/apps he has mqtt apps. Might be of benefit to you.

sharandac, 2021-09-11:
thanks for your feedback. the powermeter app is designed to work with the following [powermeter](https://github.com/sharandac/powermeter) and uses its own mqtt format. it is not designed as a general powermeter app for other powermeters.

fraschizzato, 2021-09-12:
Thanks, the part I need is the mqtt integration, following the logs I can't see any related during debug. I've placed some log/debug point but I can't see any message. Probably I'm missing something on mqtt configuration in that firmware. Is possible to have a "typical" json powermeter message to see if my tests are working with that?

sharandac, 2021-09-12:
A typical msg from the powermeter looks like this:

```json
{
     "id":"powermeter_aa5cf4",
     "all":{
            "power":"0.207",
            "PowerUnit":"kWs"
     },
     "channel0":{
            "power":"0.207",
            "voltage":"230.493",
            "current":"0.900",
            "frequence":"50.031"
     },
     "PowerUnit":"kWs",
     "VoltageUnit":"V",
     "CurrentUnit":"A",
     "FrequenceUnit":"Hz"
}
```
And by the way, don't forget to activate the autoconnect slider on page two in the powermeter settings. unfortunately a little difficult to reach, swipe up/down on the left in the settings.

closed 2021-09-19 by sharandac

fraschizzato, 2021-09-20:
Thanks a lot, it was that.