sharandac/My-TTGO-Watch #302: Build Error:  fatal error: SDL2/SDL.h: No such file or directory

hdpklm, 2021-09-17:
hello.

When I build this project with Platfomio I get this error

```
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\indev\AD_touch.o
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\indev\FT5406EE8.o
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\indev\XPT2046.o
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\indev\evdev.o
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\indev\keyboard.o
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\indev\libinput.o
.pio\libdeps\emulator_m5paper\lv_drivers\display\monitor.c:19:34: fatal error: SDL2/SDL.h: No such file or directory
 #include MONITOR_SDL_INCLUDE_PATH
                                  ^
compilation terminated.
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\indev\mouse.o
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\indev\mousewheel.o
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\win32drv\win32drv.o
Compiling .pio\build\emulator_m5paper\libab6\lv_drivers\win_drv.o
*** [.pio\build\emulator_m5paper\libab6\lv_drivers\display\monitor.o] Error 1
In file included from .pio\libdeps\emulator_m5paper\lv_drivers\indev\mouse.c:9:0:
.pio\libdeps\emulator_m5paper\lv_drivers\indev\mouse.h:36:34: fatal error: SDL2/SDL.h: No such file or directory
 #include MONITOR_SDL_INCLUDE_PATH
                                  ^
compilation terminated.
*** [.pio\build\emulator_m5paper\libab6\lv_drivers\indev\mouse.o] Error 1
================================================================= [FAILED] Took 8.37 seconds =================================================================
Environment       Status    Duration
----------------  --------  ------------
emulator_m5paper  FAILED    00:00:08.369
=========================================================== 1 failed, 0 succeeded in 00:00:08.369 =========================================================== 
The terminal process "C:\Users\Hassan Daoud\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

sharandac, 2021-09-17:
Thanks for you feedback. have you already installed the sdl2 dev lib for emulation under linux?

hdpklm, 2021-09-18:
No I am work in windows 10.
can you please add link to download the library in README

Thank you so much

hdpklm, 2021-09-18:
OK, I download the library from Adriono IDE library manager thank you so mutch

closed 2021-09-19 by sharandac