sharandac/My-TTGO-Watch #304: Display OFF or App without Standby

fraschizzato, 2021-10-01:
Hi,
I'm trying to write an app that will be always running, at a first try the battery drains in one hour...
Now I'm trying to use less hardware as possible. I need only wifi,mqtt,motor.

There's a way to turn off the display without enter Standby mode (that hangs my app)?
There's a wayh to keep my app running in standby mode? 

The concept is to continue to receive mqtt messages and vibrate at each one (without wake up screen).

Any help is apreciated,
Thanks

sharandac, 2021-10-02:
This is also possible in standby. Look at the functions

```c
bool powermgm_register_loop_cb( ... );
bool powermgm_register_cb( ... );
```

in hardware/powermgm.h. A description is in the header file. Perhaps the following issue will also help you #170

fraschizzato, 2021-10-04:
It was that, thank you!

closed 2021-10-04 by fraschizzato