sharandac/My-TTGO-Watch #316: Battery life

HerrAugust, 2021-10-30:
Hello, I notice that with screen brightness to minimum, Wifi turned off and Bluetooth turned on and connected to my smartphone, the battery lasts a nothing. Is there any way to make it last at least 1 day?

HerrAugust, 2021-11-07:
Thank you for the answer! Ok, but the battery only last around 4 hours with wifi and screen turned off and BLE turned on (and connected with an Andoid8 smartphone) all the time... Am I the only person in this situation?

sharandac, 2021-11-07:
That's really very short. Which version of the Watch are you using?

jusjak11, 2021-11-08:
Sorry to jump in, but I have exactly the same issue making the watch unusable. In my case battery lasts for about four hours with everything off, including bluetooth. The version of watch TTGO 2020 V2. Is there a setting in your firmware that I am missing, or something else? I read that there is a possibility to put all major chips, including RTC, into a deep sleep, thus making a battery to last couple days.

jusjak11, 2021-11-08:
Just to add that I measured battery capacity which is supposed to be 380mAh and is definitely above 300mAh which is OK. The real consumption of the watch is about 80mA with WiFi, BLE, screen off.

Crsarmv7l, 2021-11-09:
Hi all, I too had the 4 hour battery life. It is due to the Bluetooth "Always On" setting. After digging through the code more, I realized "Always On" isn't really necessary as the Twatch and this firmware has wake ups scheduled and the frequency is such that you wont miss anything.

It also extended my battery life substantially to almost 2 days.

jusjak11, 2021-11-10:
@Crsarmv7l  Hi, I am not sure I understand you properly. My consumption is 80mA (which makes it total 4 hours) now, with sharandac firmware on and bluetooth, WiFi, screen off. Is there a setting in sharandac firmware to go to sleep? Where is this "Always on" setting: in original firmware (the one that watch came with) or sharandac firmware (one of source files)?

HerrAugust, 2021-11-10:
I think I can reply you. The 'Always On' settings can be found in the GUI. Turn on the watch with the current firmware, go down, right, right with the swipe and then click on the Bluetooth Settings. Among the option, you'll find find the 'Always On' one, which the other user suggests to turn off. Hope it helps!

https://github.com/sharandac/My-TTGO-Watch/issues/316#issuecomment-965075107

jusjak11, 2021-11-10:
Unfortunately this setting "Bluetooth on" is set to off in my case. So, it is not the real cause for such a high consumption.

HerrAugust, 2021-11-10:
@jusjak11 are we talking about TTgo watch V3 with microphone? Maybe we are considering different versions of the watch...

jusjak11, 2021-11-10:
@HerrAugust Mine is TTGO 2020 V2 with GPS. If anybody reading this have the same version, please chime in with battery duration and settings.

Crsarmv7l, 2021-11-10:
@jusjak11 You may just have a bad battery. 

Here is where the firmware sets up sleep @HerrAugust  
https://github.com/sharandac/My-TTGO-Watch/blob/master/src/hardware/powermgm.cpp

jusjak11, 2021-11-10:
@Crsarmv7l  I measured consumption with ampermeter to be 80mA, so it is not the battery, the watch does not go to sleep for some reason. I did not analyze cpp file you attached. Does it go to sleep automatically after certain time or it has to be initialized.

HerrAugust, 2021-11-10:
@Crsarmv7l thank you the suggestion! However, this power settings are applied automatically? If not, could you explain how to enable them? I am charging my smartwatch to discover it anyways

Crsarmv7l, 2021-11-10:
@HerrAugust yes the sleeps are applied automatically in sharandacs firmware, as is the wake timings. I like to play with the firmware to suit my needs.

Definitely turn of the Always On setting and see how it works for you, because it keeps it in a different sleep level which drains battery faster.

jusjak11, 2021-11-10:
@HerrAugust All GPS settings: auto on, enable on standby, apps use gps, fake gps via ip are turn off. I also have in Battery setting:
power save: on.

Crsarmv7l, 2021-11-10:
@jusjak11 You have your gps always on with the "Enable on Standby." That keeps gps going all the time and will prevent your watch from fully sleeping and drain the battery faster

jusjak11, 2021-11-10:
@Crsarmv7l  No, that setting is off like all others related to GPS. I measured again through the power supply. Wit  everything off including screen 64.3mA@3.8V, and 67.7mA@3.8V with power save = off. Something is preventing it to go to sleep, or it has a hardware failure.

sharandac, 2021-11-10:
@jusjak11 simple take a closer look on the serial interface. Here you can see exactly which module blocks the standby mode.

by example, here blocks bluetooth the standby mode
```
[I][button.cpp:437] button_send_cb(): send button cb
[I][button.cpp:55] button_pmu_event_cb(): send BUTTON_PWR event
[I][wifictl.cpp:581] wifictl_standby(): request wifictl standby
[I][wifictl.cpp:554] wifictl_off(): request wifictl off
[I][wifictl.cpp:562] wifictl_off(): wifictl not active, prevent first run crash
[I][wifictl.cpp:588] wifictl_standby(): request wifictl standby done
[I][gui.cpp:180] gui_powermgm_event_cb(): go standby
[W][blectl.cpp:363] blectl_powermgm_event_cb(): standby blocked by "enable_on_standby" option
[I][callback.cpp:232] callback_send(): cb powermgm blectl returns false
[I][sdcard.cpp:115] sdcard_powermgm_event_cb(): go standby
[I][framebuffer.cpp:171] framebuffer_powermgm_event_cb(): go standby, refresh framebuffer
[I][display.cpp:213] display_standby(): go standby
[I][motion.cpp:326] bma_standby(): go standby
[I][touch.cpp:272] touch_powermgm_event_cb(): go standby
[I][rtcctl.cpp:270] rtcctl_powermgm_event_cb(): go standby
[I][timesync.cpp:111] timesync_powermgm_event_cb(): go standby
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /blectl.json
[I][sound.cpp:151] sound_powermgm_event_cb(): go standby
[I][gpsctl.cpp:297] gpsctl_powermgm_event_cb(): go standby
[I][pmu.cpp:543] pmu_standby(): go standby, enable 3000mV standby voltage
[I][powermgm.cpp:201] powermgm_loop(): Free heap: 117620
[I][powermgm.cpp:202] powermgm_loop(): Free PSRAM heap: 3667408
[I][powermgm.cpp:203] powermgm_loop(): uptime: 571716
[I][powermgm.cpp:254] powermgm_loop(): go standby blocked
[I][powermgm.cpp:272] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 80/40MHz with light sleep
```

jusjak11, 2021-11-10:
@sharandac Here is the extract from the serial port

[I][powermgm.cpp:114] powermgm_loop(): go silence wakeup
[I][pmu.cpp:588] pmu_wakeup(): go wakeup, enable 3000mV voltage
[I][sdcard.cpp:85] sdcard_powermgm_event_cb(): go silence wakeup
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:741] sdcard_mount(): f_mount failed 0x(3)
[E][sdcard.cpp:88] sdcard_powermgm_event_cb(): SD Card Mount Failed
[I][framebuffer.cpp:176] framebuffer_powermgm_event_cb(): go wakeup
[I][display.cpp:249] display_wakeup(): go silence wakeup
[I][motion.cpp:360] bma_wakeup(): go wakeup
[I][touch.cpp:290] touch_powermgm_event_cb(): go silence wakeup
[I][rtcctl.cpp:287] rtcctl_powermgm_event_cb(): go silence wakeup
[I][timesync.cpp:126] timesync_powermgm_event_cb(): go silence wakeup
[I][timesync.cpp:280] timesyncToSystem(): TZ rule: CET-1CEST,M3.5.0,M10.5.0/3
[I][gpsctl.cpp:306] gpsctl_powermgm_event_cb(): go silence wakeup
[I][blectl.cpp:377] blectl_powermgm_event_cb(): go silence wakeup
[I][gui.cpp:201] gui_powermgm_event_cb(): go silence wakeup
[I][powermgm.cpp:134] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 160/80MHz with light sleep
[I][powermgm.cpp:172] powermgm_loop(): Free heap: 113268
[I][powermgm.cpp:173] powermgm_loop(): Free PSRAM heap: 3438736
[I][powermgm.cpp:174] powermgm_loop(): uptime: 6280
[I][wifictl.cpp:580] wifictl_standby(): request wifictl standby
[I][wifictl.cpp:553] wifictl_off(): request wifictl off
[I][wifictl.cpp:561] wifictl_off(): wifictl not active, prevent first run crash
[I][wifictl.cpp:587] wifictl_standby(): request wifictl standby done
[I][gui.cpp:180] gui_powermgm_event_cb(): go standby
[I][blectl.cpp:370] blectl_powermgm_event_cb(): go standby
[I][sdcard.cpp:94] sdcard_powermgm_event_cb(): go standby
[I][framebuffer.cpp:171] framebuffer_powermgm_event_cb(): go standby, refresh framebuffer
[I][display.cpp:213] display_standby(): go standby
[I][motion.cpp:326] bma_standby(): go standby
[I][touch.cpp:272] touch_powermgm_event_cb(): go standby
[I][rtcctl.cpp:270] rtcctl_powermgm_event_cb(): go standby
[I][timesync.cpp:111] timesync_powermgm_event_cb(): go standby
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /blectl.json
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /blectl.json
[I][gpsctl.cpp:297] gpsctl_powermgm_event_cb(): go standby
[I][pmu.cpp:526] pmu_standby(): enable silence wakeup timer, 1min
[I][pmu.cpp:539] pmu_standby(): go standby, enable 2800mV standby voltage
[I][powermgm.cpp:201] powermgm_loop(): Free heap: 113316
[I][powermgm.cpp:202] powermgm_loop(): Free PSRAM heap: 3438932
[I][powermgm.cpp:203] powermgm_loop(): uptime: 6296
[I][powermgm.cpp:207] powermgm_loop(): go standby
[I][powermgm.cpp:219] powermgm_loop(): CPU speed = 80MHz, start light sleep
[I][powermgm.cpp:244] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep

What does it tell you?

jusjak11, 2021-11-12:
As I can see it goes to light sleep and consumption is still 63mA. How is that possible? What is taking this 63mA?

jusjak11, 2021-11-12:
I can see that consumption goes down to 33mA briefly and than back to 63mA, every minute, so it tries to go to sleep but something wakes it up again.

sharandac, 2021-11-12:
@jusjak11 It would be interesting to see the pmu.json.There, the Watch is set to wake up every minute when USB is connected.  If USB is not connected, every 5 minutes. The Watch then goes back to sleep after the display timeout, default 15s.
And by the way, it's a bad idea to measure the current when USB is connected, that takes round about 19mA extra.
First, maybe try turning off "silence wakeup" and see if it gets better.

```json
{
  "silence_wakeup": true,
  "silence_wakeup_interval": 5,
  "silence_wakeup_interval_vbplug": 1,
  "experimental_power_save": false,
  "normal_voltage": 3300,
  "normal_power_save_voltage": 3000,
  "experimental_normal_voltage": 3000,
  "experimental_power_save_voltage": 2800,
  "compute_percent": false,
  "high_charging_target_voltage": false,
  "designed_battery_cap": 300,
  "pmu_logging": false
}
```

jusjak11, 2021-11-13:
@sharandac Where is pmu.json file? Sorry I am new in this.

jusjak11, 2021-11-14:
Measured consumption again to verify this 5 minute timer. The consumption is, without USB connected, screen off, BLE off, WIFi off  63mA @ 3.8V and every 5 minutes it goes down to 33mA briefly and than back to 63mA, which means it does try to go to sleep but something prevents it, something which cannot be seen over the serial port. Notice that serial port is not connected so 19mA does not apply.

sharandac, 2021-11-14:
It's starting to sound to me like we're chasing a phantom that doesn't exist. The measurement results are so contradictory that I doubt the measurement.
We have seen on the serial interface that the Watch goes to sleep, i.e. saves power. Okay, that's done. The software is working properly.
But if we interpret your measurement results, they say exactly the opposite. What does the measurement setup look like?

jusjak11, 2021-11-14:
@sharandac  You don't have to have any special measurement setup when the watch battery in normal use lasts 4 to 4.5 hours. So, if you have good battery you know that consumption is real. Than if you measure the current to be 63mA you know that the battery is good and watch really consumes it. 
The test setup is: watch is powered with power supply set to 3.8V, directly to input pins where battery is connected. Battery is disconnected. Ampermeter is in series to power supply. The USB cable is not connected, thus serial communication is not monitored (the reason is I don't want to have double supply: from USB and battery input). In bootup time consumption goes to 110mA. After bootup, with screen on : 85mA. After default 15s screen goes off: 63mA. After 5min. there is a glitch: consumption goes briefly to 33ma and right after it back to 66mA. And this repeats every 5 minutes.

When watch powered from USB I do not measure current because I did not make split cable to connect ampermeter but monitor serial port. It behaves different than on battery supply. For some reason, sometimes it goes to sleep mode and stays there for a minute and so on, but sometimes right after went to sleep something wakes it up right after it. 

So I managed to recreate error here, but I do not know what is triggering it, and why sometimes it happens, simetimes not as shown on serial port. It looks like the reason is in type of power: USB or battery, and battery voltage, how it boots up. I am just giving some hints.

This is seen on serial port when watch is waken up right after going to sleep:

[I][powermgm.cpp:207] powermgm_loop(): go standby
[I][powermgm.cpp:219] powermgm_loop(): CPU speed = 80MHz, start light sleep
[I][powermgm.cpp:244] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:143] powermgm_loop(): go wakeup
[I][pmu.cpp:588] pmu_wakeup(): go wakeup, enable 3000mV voltage
[I][sdcard.cpp:99] sdcard_powermgm_event_cb(): go wakeup
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:741] sdcard_mount(): f_mount failed 0x(3)
[E][sdcard.cpp:102] sdcard_powermgm_event_cb(): SD Card Mount Failed
[I][framebuffer.cpp:174] framebuffer_powermgm_event_cb(): go wakeup
[I][display.cpp:280] display_wakeup(): go wakeup
[I][motion.cpp:360] bma_wakeup(): go wakeup
[I][touch.cpp:279] touch_powermgm_event_cb(): go wakeup
[I][rtcctl.cpp:285] rtcctl_powermgm_event_cb(): go wakeup
[I][timesync.cpp:119] timesync_powermgm_event_cb(): go wakeup
[I][timesync.cpp:280] timesyncToSystem(): TZ rule: CET-1CEST,M3.5.0,M10.5.0/3
[I][gpsctl.cpp:302] gpsctl_powermgm_event_cb(): go wakeup
[I][blectl.cpp:374] blectl_powermgm_event_cb(): go wakeup
[I][gui.cpp:191] gui_powermgm_event_cb(): go wakeup
[I][powermgm.cpp:163] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:172] powermgm_loop(): Free heap: 114228
[I][powermgm.cpp:173] powermgm_loop(): Free PSRAM heap: 3439108
[I][powermgm.cpp:174] powermgm_loop(): uptime: 147
[I][wifictl.cpp:580] wifictl_standby(): request wifictl standby
[I][wifictl.cpp:553] wifictl_off(): request wifictl off
[I][wifictl.cpp:561] wifictl_off(): wifictl not active, prevent first run crash
[I][wifictl.cpp:587] wifictl_standby(): request wifictl standby done
[I][gui.cpp:180] gui_powermgm_event_cb(): go standby
[I][blectl.cpp:370] blectl_powermgm_event_cb(): go standby
[I][sdcard.cpp:94] sdcard_powermgm_event_cb(): go standby
[I][framebuffer.cpp:171] framebuffer_powermgm_event_cb(): go standby, refresh framebuffer
[I][display.cpp:213] display_standby(): go standby
[I][motion.cpp:326] bma_standby(): go standby
[I][touch.cpp:272] touch_powermgm_event_cb(): go standby
[I][rtcctl.cpp:270] rtcctl_powermgm_event_cb(): go standby
[I][timesync.cpp:111] timesync_powermgm_event_cb(): go standby
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /blectl.json
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /blectl.json
[I][gpsctl.cpp:297] gpsctl_powermgm_event_cb(): go standby
[I][pmu.cpp:526] pmu_standby(): enable silence wakeup timer, 1min
[I][pmu.cpp:539] pmu_standby(): go standby, enable 2800mV standby voltage
[I][powermgm.cpp:201] powermgm_loop(): Free heap: 114276
[I][powermgm.cpp:202] powermgm_loop(): Free PSRAM heap: 3439208
[I][powermgm.cpp:203] powermgm_loop(): uptime: 163
[I][powermgm.cpp:207] powermgm_loop(): go standby
[I][powermgm.cpp:219] powermgm_loop(): CPU speed = 80MHz, start light sleep
[I][powermgm.cpp:244] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:143] powermgm_loop(): go wakeup
[I][pmu.cpp:588] pmu_wakeup(): go wakeup, enable 3000mV voltage
[I][sdcard.cpp:99] sdcard_powermgm_event_cb(): go wakeup
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:741] sdcard_mount(): f_mount failed 0x(3)
[E][sdcard.cpp:102] sdcard_powermgm_event_cb(): SD Card Mount Failed
[I][framebuffer.cpp:174] framebuffer_powermgm_event_cb(): go wakeup
[I][display.cpp:280] display_wakeup(): go wakeup
[I][motion.cpp:360] bma_wakeup(): go wakeup
[I][touch.cpp:279] touch_powermgm_event_cb(): go wakeup
[I][rtcctl.cpp:285] rtcctl_powermgm_event_cb(): go wakeup
[I][timesync.cpp:119] timesync_powermgm_event_cb(): go wakeup
[I][timesync.cpp:280] timesyncToSystem(): TZ rule: CET-1CEST,M3.5.0,M10.5.0/3
[... cut ...]
[I][powermgm.cpp:202] powermgm_loop(): Free PSRAM heap: 3439084
[I][powermgm.cpp:203] powermgm_loop(): uptime: 183
[I][powermgm.cpp:207] powermgm_loop(): go standby
[I][powermgm.cpp:219] powermgm_loop(): CPU speed = 80MHz, start light sleep
[I][powermgm.cpp:244] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:143] powermgm_loop(): go wakeup
[I][pmu.cpp:588] pmu_wakeup(): go wakeup, enable 3000mV voltage
[I][sdcard.cpp:99] sdcard_powermgm_event_cb(): go wakeup

sharandac, 2021-11-14:
The setup doesn't sound bad. What surprises me about the log files is the too short interval between standby and wakeup ... 4 seconds? So there must be another reason why the Watch wakes up again after such a short time. The setting is 1min.
 The interesting thing is that it doesn't go into silence wakeup but wakes up properly. Does this also happen if you only supply the clock via USB without a battery inserted?
```
[I][gui.cpp:191] gui_powermgm_event_cb(): go wakeup
[I][powermgm.cpp:163] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:172] powermgm_loop(): Free heap: 114228
[I][powermgm.cpp:173] powermgm_loop(): Free PSRAM heap: 3439108
[I][powermgm.cpp:174] powermgm_loop(): uptime: 147
```
```
[I][pmu.cpp:526] pmu_standby(): enable silence wakeup timer, 1min
[I][pmu.cpp:539] pmu_standby(): go standby, enable 2800mV standby voltage
[I][powermgm.cpp:201] powermgm_loop(): Free heap: 114276
[I][powermgm.cpp:202] powermgm_loop(): Free PSRAM heap: 3439208
[I][powermgm.cpp:203] powermgm_loop(): uptime: 163
```
```
[I][gui.cpp:191] gui_powermgm_event_cb(): go wakeup
[I][powermgm.cpp:163] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:172] powermgm_loop(): Free heap: 114180
[I][powermgm.cpp:173] powermgm_loop(): Free PSRAM heap: 3439084
[I][powermgm.cpp:174] powermgm_loop(): uptime: 167
```
```
[I][pmu.cpp:526] pmu_standby(): enable silence wakeup timer, 1min
[I][pmu.cpp:539] pmu_standby(): go standby, enable 2800mV standby voltage
[I][powermgm.cpp:201] powermgm_loop(): Free heap: 114228
[I][powermgm.cpp:202] powermgm_loop(): Free PSRAM heap: 3439084
[I][powermgm.cpp:203] powermgm_loop(): uptime: 183
```

jusjak11, 2021-11-14:
@sharandac With USB only power and battery disconnected it goes to sleep and stays there 1 minute as it is supposed to be. Here is log file:

[I][powermgm.cpp:114] powermgm_loop(): go silence wakeup
[I][pmu.cpp:588] pmu_wakeup(): go wakeup, enable 3000mV voltage
[I][sdcard.cpp:85] sdcard_powermgm_event_cb(): go silence wakeup
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:741] sdcard_mount(): f_mount failed 0x(3)
[E][sdcard.cpp:88] sdcard_powermgm_event_cb(): SD Card Mount Failed
[I][framebuffer.cpp:176] framebuffer_powermgm_event_cb(): go wakeup
[I][display.cpp:249] display_wakeup(): go silence wakeup
[I][motion.cpp:360] bma_wakeup(): go wakeup
[I][touch.cpp:290] touch_powermgm_event_cb(): go silence wakeup
[I][rtcctl.cpp:287] rtcctl_powermgm_event_cb(): go silence wakeup
[I][timesync.cpp:126] timesync_powermgm_event_cb(): go silence wakeup
[I][timesync.cpp:280] timesyncToSystem(): TZ rule: CET-1CEST,M3.5.0,M10.5.0/3
[I][gpsctl.cpp:306] gpsctl_powermgm_event_cb(): go silence wakeup
[I][blectl.cpp:377] blectl_powermgm_event_cb(): go silence wakeup
[I][gui.cpp:201] gui_powermgm_event_cb(): go silence wakeup
[I][powermgm.cpp:134] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 160/80MHz with light sleep
[I][powermgm.cpp:172] powermgm_loop(): Free heap: 114516
[I][powermgm.cpp:173] powermgm_loop(): Free PSRAM heap: 3439192
[I][powermgm.cpp:174] powermgm_loop(): uptime: 86
[I][wifictl.cpp:580] wifictl_standby(): request wifictl standby
[I][wifictl.cpp:553] wifictl_off(): request wifictl off
[I][wifictl.cpp:561] wifictl_off(): wifictl not active, prevent first run crash
[I][wifictl.cpp:587] wifictl_standby(): request wifictl standby done
[I][gui.cpp:180] gui_powermgm_event_cb(): go standby
[I][blectl.cpp:370] blectl_powermgm_event_cb(): go standby
[I][sdcard.cpp:94] sdcard_powermgm_event_cb(): go standby
[I][framebuffer.cpp:171] framebuffer_powermgm_event_cb(): go standby, refresh framebuffer
[I][display.cpp:213] display_standby(): go standby
[I][motion.cpp:326] bma_standby(): go standby
[I][touch.cpp:272] touch_powermgm_event_cb(): go standby
[I][rtcctl.cpp:270] rtcctl_powermgm_event_cb(): go standby
[I][timesync.cpp:111] timesync_powermgm_event_cb(): go standby
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /blectl.json
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /blectl.json
[I][gpsctl.cpp:297] gpsctl_powermgm_event_cb(): go standby
[I][pmu.cpp:526] pmu_standby(): enable silence wakeup timer, 1min
[I][pmu.cpp:539] pmu_standby(): go standby, enable 2800mV standby voltage
[I][powermgm.cpp:201] powermgm_loop(): Free heap: 114564
[I][powermgm.cpp:202] powermgm_loop(): Free PSRAM heap: 3439400
[I][powermgm.cpp:203] powermgm_loop(): uptime: 102
[I][powermgm.cpp:207] powermgm_loop(): go standby
[I][powermgm.cpp:219] powermgm_loop(): CPU speed = 80MHz, start light sleep
[I][powermgm.cpp:244] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep
[I][powermgm.cpp:114] powermgm_loop(): go silence wakeup
[I][pmu.cpp:588] pmu_wakeup(): go wakeup, enable 3000mV voltage
[I][sdcard.cpp:85] sdcard_powermgm_event_cb(): go silence wakeup
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:741] sdcard_mount(): f_mount failed 0x(3)
[E][sdcard.cpp:88] sdcard_powermgm_event_cb(): SD Card Mount Failed
[I][framebuffer.cpp:176] framebuffer_powermgm_event_cb(): go wakeup
[I][display.cpp:249] display_wakeup(): go silence wakeup
[I][motion.cpp:360] bma_wakeup(): go wakeup
[I][touch.cpp:290] touch_powermgm_event_cb(): go silence wakeup
[I][rtcctl.cpp:287] rtcctl_powermgm_event_cb(): go silence wakeup
[I][timesync.cpp:126] timesync_powermgm_event_cb(): go silence wakeup
[I][timesync.cpp:280] timesyncToSystem(): TZ rule: CET-1CEST,M3.5.0,M10.5.0/3
[I][gpsctl.cpp:306] gpsctl_powermgm_event_cb(): go silence wakeup
[I][blectl.cpp:377] blectl_powermgm_event_cb(): go silence wakeup
[I][gui.cpp:201] gui_powermgm_event_cb(): go silence wakeup
[I][powermgm.cpp:134] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 160/80MHz with light sleep
[I][powermgm.cpp:172] powermgm_loop(): Free heap: 114468
[I][powermgm.cpp:173] powermgm_loop(): Free PSRAM heap: 3439188
[I][powermgm.cpp:174] powermgm_loop(): uptime: 162
[I][wifictl.cpp:580] wifictl_standby(): request wifictl standby
[I][wifictl.cpp:553] wifictl_off(): request wifictl off
[I][wifictl.cpp:561] wifictl_off(): wifictl not active, prevent first run crash
[I][wifictl.cpp:587] wifictl_standby(): request wifictl standby done
[... 41 lines cut ...]
[I][powermgm.cpp:174] powermgm_loop(): uptime: 198
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /blectl.json
[I][wifictl.cpp:580] wifictl_standby(): request wifictl standby
[I][wifictl.cpp:553] wifictl_off(): request wifictl off
[I][wifictl.cpp:561] wifictl_off(): wifictl not active, prevent first run crash
[I][wifictl.cpp:587] wifictl_standby(): request wifictl standby done
[I][gui.cpp:180] gui_powermgm_event_cb(): go standby
[I][blectl.cpp:370] blectl_powermgm_event_cb(): go standby
[I][sdcard.cpp:94] sdcard_powermgm_event_cb(): go standby
[I][framebuffer.cpp:171] framebuffer_powermgm_event_cb(): go standby, refresh framebuffer
[I][display.cpp:213] display_standby(): go standby
[I][motion.cpp:326] bma_standby(): go standby
[I][touch.cpp:272] touch_powermgm_event_cb(): go standby
[... cut ...]
[I][pmu.cpp:539] pmu_standby(): go standby, enable 2800mV standby voltage
[I][powermgm.cpp:201] powermgm_loop(): Free heap: 114468
[I][powermgm.cpp:202] powermgm_loop(): Free PSRAM heap: 3439128
[I][powermgm.cpp:203] powermgm_loop(): uptime: 215
[I][powermgm.cpp:207] powermgm_loop(): go standby
[I][powermgm.cpp:219] powermgm_loop(): CPU speed = 80MHz, start light sleep
[I][powermgm.cpp:244] powermgm_loop(): custom arduino-esp32 framework detected, enable PM/DFS support, 240/80MHz with light sleep

jusjak11, 2021-11-14:
Right now I tried it on USB power but battery connected. It worked properly i.e. goes to sleep and sleeps for 1 minute. Maybe it depends on state of charge of battery; battery was empty?

jusjak11, 2021-11-14:
@sharandac Just to emphasize one more time, if you haven't read whole thread, my watch is TTGO 2020 V2, the one with GPS.

closed 2022-01-05 by sharandac