sharandac/My-TTGO-Watch #319: Error trying to compile V1 2020 watch

kaman112, 2021-12-03:
Hi,
I followed your youtube tutorial however I cannot get it to compile successfully for V1 it comes up with the following messages:
Linking .pio\build\t-watch2020-v1\firmware.elf
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\t-watch2020-v1\firmware.elf] Error 1

I did successfully compile V2 as a test and tried V3 but it also failed

Any ideas?

liamharrison, 2021-12-04:
I got it to compile by removing some of the apps from main.cpp and removing the src dirs for the apps I removed. I think the command line for g++ is too long, this might be nonsense.

liamharrison, 2021-12-04:
ignore that ^^^^^, even once you get it to build it makes the watch reboot-loop

commit 5b53a24 builds with no problems

Crsarmv7l, 2021-12-07:
> even if you get the current master to build it makes (my) the watch reboot-loop
> 
> commit [5b53a24](https://github.com/sharandac/My-TTGO-Watch/commit/5b53a24ac3b47aca77a2e1fe4357286d180700d8) (a few days before) builds with no problems
> 
> more edit: ^^^^ this also reboot loops had to go to some mid april commit

Weird. I only got my watch in Aug/Sep and it built fine then with no bootloop. Built a more recent one yesterday (Oct I think), and it reboots with serial monitor, something it didn't do before. I haven't tested extensively yet, but there doesn't seem to be a bootloop just the one reboot if I start or stop serial monitor.

sharandac, 2021-12-10:
PlatformIO have some strange bugs. under windows. It sounds like this ... too long and to much files for the command line to compile. It is a know bug. and the question is, witch OS is in use?

liamharrison, 2021-12-11:
> PlatformIO have some strange bugs. under windows. It sounds like this ... too long and to much files for the command line to compile. It is a know bug. and the question is, witch OS is in use?

It was windows where it failed (if I remove files from the build to shorten the cmdline it completes). same build completed on linux. (still resulted in a non-functioning watch firmware, but that's a different matter)

Crsarmv7l, 2021-12-12:
>To get the FW to build on Windows, comment out some apps you don't need in main.c

Ah that explains why I dont have the issue. I always pair down significantly

kaman112, 2021-12-12:
Hi all,

I have tried again with Linux and I get similar issues for both it and Windows the xtensa issues seem to have gone but 
 I consistently seem to have some issue with lvgl.  
Early in the build it flags "WarningCould not find the package with "lvgl/lvgl @  = 7.7.2 requirements for your system 

In file included from src/app/FindPhone/FindPhone.cpp:25:0:
src/config.h:25:22: fatal error: lvgl.h: No such file or directory

Do I need to add a lvgl.h file to a particular folder that currently does not contain it?

Hopefully this makes sense.

liamharrison, 2021-12-12:
> Hi all,
>
> I have tried again with Linux and I get similar issues for both it and Windows the xtensa issues seem to have gone but I consistently seem to have some issue with lvgl. Early in the build [...]

You can use platformio to install lvgl...
`pio lib install "lvgl/lvgl@^7.11.0"` (I've chosen 7.11.0 because I think the latest version doesn't work)

MoorPoor, 2021-12-12:
Thanks.
Copy & pasting [7.11.0 from git ](https://github.com/lvgl/lvgl/releases/tag/v7.11.0) in the lib folder (.pio/libdeps/t-watch2020-vXY) worked als for me.

kaman112, 2021-12-21:
Hi all,

I have successfully managed to build V1 with version 7.11.0 and using a linux OS to avoid the xtensa issues which is great, however I now face the same problems mentioned in this thread https://github.com/sharandac/My-TTGO-Watch/issues/322

I also tried using an earlier version (from around July to test and it still seems to have the continuous reboot issue.

So I guess I should close this thread as 7.11.0 and linux seems to solve these particular issues, thanks for all the assistance.

closed 2021-12-21 by kaman112