sharandac/My-TTGO-Watch #320: FTP unable to connect

nieldk, 2021-12-09:
I cant get the  2020v3 to connect to FTP.
Using the latest OTA update, and trying the 'default' username password found in the source for the FTP app (TTWatch/password)

But it disconnects with user not found (after actually sending a syntax error)

sharandac, 2021-12-10:
Thanks for your feedback. One this is the use of passive ftp, not all client support it. Here is a log for a ftp session with the last version:

```
(base) sharan@warbird:~$ pftp 192.168.2.26
Connected to 192.168.2.26.
220--- Welcome to FTP for ESP8266/ESP32 ---
220---   By David Paiva   ---
220 --   Version FTP-2017-10-18   --
Name (192.168.2.26:sharan): TTWatch
331 OK. Password required
Password:
230 OK.
ftp> ls
227 Entering Passive Mode (192,168,2,26,195,89).
150 Accepted data connection
+r,s12288
,	/calendar.db
+r,s136
,	/gpsctl.json
+r,s102
,	/bma.json
+r,s419
,	/pmu.json
+r,s105
,	/alaram.json
+r,s133
,	/weather.json
+r,s182
,	/timesync.json
+r,s128
,	/update.json
+r,s28874
,	/watchface.tar.gz
+r,s31271
,	/watchface/watchface_dial.png
+r,s90
,	/rtcctr.json
+r,s2749
,	/watchface/watchface_theme.json
+r,s161
,	/watchface.json
+r,s1247
,	/wificfg.json
+r,s1267
,	/blectl.json
+r,s194
,	/display.json
226 16 matches total
ftp> quit
221 Goodbye
```

nieldk, 2021-12-10:
Excellent. That was it. Windoze tries active FTP, which fails. Closing the issue.

closed 2021-12-10 by nieldk