sharandac/My-TTGO-Watch #322: T-Watch-v2 continously rebooting after uploading new version

Jan-NiklasB, 2021-12-11:
Hey there,
I uploaded the actual version (cloned on 2021-12-11, Stamp: f1a3344afc52b12df488e3bc4ead0654b69e9a42) on my factory new T-Watch V2.
After flashing the watch continously reboots.

I had to pull lvgl 7.11 instead of 7.8 bvecaouse the 7.8 isn't available.

Some help would be appreciated.

pruttelherrie, 2021-12-15:
Having the same issue here on TWatch 2020 v1: "... gui" -> screen dims -> hedgehog.
This watch has been successfully running an older version (from march?) before being updated.

MoorPoor, 2021-12-15:
hei
Are you uploading with platformIO / visual studio or with arduino IDE?

pruttelherrie, 2021-12-15:
I'm using VSCode + PIO on Linux.

MoorPoor, 2021-12-15:
same setup here.
strange... You added the lvgl lib like mentioned [here](https://github.com/sharandac/My-TTGO-Watch/issues/319#issuecomment-991878952)?

if that is not helping for now, try watchface-ng or another older brench. they should work

pruttelherrie, 2021-12-15:
No, I had     lvgl/lvgl@^7.7.2 in Platformio.ini, which pulled lvgl 7.11

I tried to add the lvgl lib like mentioned in the other issue, still rebooting.
Serial terminal shows:

```
[I][gui.cpp:99] gui_setup(): note tile setup
[I][gui.cpp:101] gui_setup(): setup tile setup
[I][setup_tile.cpp:101] setup_tile_setup(): setup tile finish
[I][gui.cpp:106] gui_setup(): statusbar setup
[I][callback.cpp:92] callback_init(): init callback_t structure success for: bma
[I][callback.cpp:92] callback_init(): init callback_t structure success for: display
[I][callback.cpp:92] callback_init(): init callback_t structure success for: sound
[I][gui.cpp:108] gui_setup(): quickbar setup
[I][gui.cpp:110] gui_setup(): keyboard setup
[I][gui.cpp:112] gui_setup(): num keyboard setup
[I][widget_factory.cpp:536] wf_get_settings_header_title(): header_title found
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /update.json
[I][callback.cpp:92] callback_init(): init callback_t structure success for: http ota
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060930  A0      : 0x80170e81  A1      : 0x3ffd8e90  
A2      : 0x3ffbe0cc  A3      : 0x3ffd8eb4  A4      : 0x00000054  A5      : 0x00000065  
A6      : 0xffffffff  A7      : 0x00000055  A8      : 0x80306d2c  A9      : 0x000000ff  
A10     : 0x3ffbe0cc  A11     : 0x3ffd8eb4  A12     : 0x00000054  A13     : 0x00000065  
A14     : 0x3f818738  A15     : 0x3f818738  SAR     : 0x00000001  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4030694c  LEND    : 0x40306959  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x00000000:0x3ffd8e90 0x40170e7e:0x3ffd8eb0 0x40173b12:0x3ffd8ee0 0x40173e61:0x3ffd8f20 0x401827aa:0x3ffd8f60 0x40182d9a:0x3ffd9000 0x4018313d:0x3ffd9020 0x400e847c:0x3ffd9040 0x400e887b:0x3ffd9060 0x400e49e3:0x3ffd9090 0x40106d1a:0x3ffd90b0 0x401bace6:0x3ffd90d0 0x400953da:0x3ffd90f0
  #0  0x00000000:0x3ffd8e90 in ?? ??:0
  #1  0x40170e7e:0x3ffd8eb0 in lv_font_get_glyph_width at .pio/libdeps/t-watch2020-v1/TTGO TWatch Library/src/lvgl/src/lv_font/lv_font.c:78
  #2  0x40173b12:0x3ffd8ee0 in _lv_txt_get_width at .pio/libdeps/t-watch2020-v1/TTGO TWatch Library/src/lvgl/src/lv_misc/lv_txt.c:605
  #3  0x40173e61:0x3ffd8f20 in _lv_txt_get_size at .pio/libdeps/t-watch2020-v1/TTGO TWatch Library/src/lvgl/src/lv_misc/lv_txt.c:605
  #4  0x401827aa:0x3ffd8f60 in lv_label_refr_text at .pio/libdeps/t-watch2020-v1/TTGO TWatch Library/src/lvgl/src/lv_widgets/lv_label.c:395
  #5  0x40182d9a:0x3ffd9000 in lv_label_set_text at .pio/libdeps/t-watch2020-v1/TTGO TWatch Library/src/lvgl/src/lv_widgets/lv_label.c:395
  #6  0x4018313d:0x3ffd9020 in lv_label_create at .pio/libdeps/t-watch2020-v1/TTGO TWatch Library/src/lvgl/src/lv_widgets/lv_label.c:395
  #7  0x400e847c:0x3ffd9040 in bluetooth_pairing_tile_setup() at src/gui/mainbar/setup_tile/bluetooth_settings/bluetooth_pairing.cpp:76
  #8  0x400e887b:0x3ffd9060 in bluetooth_settings_tile_setup() at src/gui/mainbar/setup_tile/bluetooth_settings/bluetooth_settings.cpp:137
  #9  0x400e49e3:0x3ffd9090 in gui_setup() at src/gui/gui.cpp:131
  #10 0x40106d1a:0x3ffd90b0 in setup() at src/main.cpp:48
  #11 0x401bace6:0x3ffd90d0 in loopTask(void*) at /home/iwan/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
  #12 0x400953da:0x3ffd90f0 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

Jan-NiklasB, 2021-12-15:
Sorry, for some reason I wasn't notified about the new posts....

@MoorPoor I for my part used PlatformIO as suggested (tested under Win7 & Linux Mint)

pruttelherrie, 2021-12-15:
Ok, while trying to re-build (which takes some time on my laptop ;) I connected the watch again and after a couple of reboots it started to act normal?!? 
For now it seems to work ok. (please note that I *did* copy the lvgl lib to .pio/libdeps/t-watch2020-v1)

Weird.

Jan-NiklasB, 2021-12-17:
I didn't copied the lvgl but pulled it by PaltformIO...
The watch kept restarting for 10 mins or so.... so in consequence I disconnected the battery.... Also the version flashed by factory was unstable and rebooted after a short while, which was the reason for me to flash the newest version....

closed 2022-01-05 by sharandac