sharandac/My-TTGO-Watch #323: IR App help

Crsarmv7l, 2021-12-12:
Hi,

Looking for some insight about your IR Remote app

I have 3 TV's in my house, and for simplicity sake I would like to create one remote that sends 3 commands (one for each tv) with each button press. Think the old school tv-b-gone. That would simplify the interface/remotes, reducing it to one remote on screen, which should work on 3 different tvs. Not the most elegant, but effective.

The remote file would look like this:

{ "Remote": [{ "Power": { "m": 7, "hex": "E0E040BF" "m": tv2, "hex": "tv2 hex" "m": tv3, "hex": "tv3 hex" }, "Ch+": { "m": 7, "hex": "E0E048B7" "m": tv2, "hex": "tv2 hex" "m": tv3, "hex": "tv3 hex" }, "Ch-": { "m": 7, "hex": "E0E008F7" "m": tv2, "hex": "tv2 hex" "m": tv3, "hex": "tv3 hex" } ......... } ], "defBtnHeight": 33, "defBtnWidth": 65, "defSpacing": 2 }

Obviously the "m" and hex values would be replaced by the appropriate values.

It doesn't seem to work with the ir app though. Any tips?

d03n3rfr1tz3, 2021-12-13:
Took me a bit longer then usual, but I answered this here: https://github.com/d03n3rfr1tz3/TTGO.T-Watch.2020/issues/15

Regarding the feature for defining multiple commands for one button, I will probably implement that in the near future. Depends a little bit on when I have time and motivation to do it 😉
Of course I will create a PR when it's done to bring it to the real firmware (sharandacs one).

Crsarmv7l, 2021-12-14:
> Took me a bit longer then usual, but I answered this here: [d03n3rfr1tz3#15](https://github.com/d03n3rfr1tz3/TTGO.T-Watch.2020/issues/15)
> 
> Regarding the feature for defining multiple commands for one button, I will probably implement that in the near future. Depends a little bit on when I have time and motivation to do it 😉 Of course I will create a PR when it's done to bring it to the real firmware (sharandacs one).

Thank you. I created the issue on your github before I found your writeup and discovered that the IR app was already in the firmware when you forked, so I figured I would put it here.

I look forward to the implementation! 

IR, opensource and the BLE UART, are the main reasons I got this. I was able to get it working with Sailfish OS, now just need IR to finish it off.

Crsarmv7l, 2021-12-17:
d03n3rfr1tz3 is awesome.

closed 2021-12-17 by Crsarmv7l

d03n3rfr1tz3, 2021-12-17:
do not celebrate to early, it' just merged into experimental currently. it "works on my machine" 😉