sharandac/My-TTGO-Watch #328: xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory *** [.pio\build\t-watch2020-v1\firmware.elf] Error 1

defynite, 2022-01-04:
received when trying to build or upload

sharandac, 2022-01-05:
Thanks for your feedback. It sounds like PIO is not right installed. https://github.com/sharandac/My-TTGO-Watch#install

d03n3rfr1tz3, 2022-01-05:
It's also possible, that this is the typical windows-bug with the long command line that packs everything together. I have this problem since a few month on my fork, because of all the apps I have. It's just the number of files combined with the length of the paths and filenames that increase the commandline to a point, where windows says "nope, I'm too old for this sh*t".

If that is the case and you still want to compile on windows, you have to shorten paths. For example, if your project folder is in "C:\Users\someusername\Documents\Visual Studio Code\My-TTGO-Watch" it could already be enough to move it to "C:\My-TTGO-Watch". Alternativly you can change your build folder, which is by default in your project folder under ".pio", which I did, so that it builds into "C:\pio" instead of "C:\My-TTGO-Watch\\.pio"

If that somehow is not enough, which it was in my case, you have to shorten the paths more. The only way to shorten it more is to rename your desired environment name (in my case "t-watch2020-v1") to something shorter (for example 0v1), because it will be part of the path of every file in the commandline later. Everything combined from above should yield to a very short build folder (for example "C:\pio\0v1") which drastically shortens the resulting commandline that packs everything together.

jan-marten, 2022-01-13:
I am having the same problem on multiple systems (even on a freshly installed Windows 10 VM).
Source code is located in "C:\pio\My-TTGO-Watch".

`xtensa-esp32-elf-g++ @C:\pio\My-TTGO-Watch\.pio\build\t-watch2020-v3\tmpedq_pa_w.tmp
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\t-watch2020-v3\firmware.elf] Error 1`

Attached is the full log.

[My-TTGO-Watch verbose build.log](https://github.com/sharandac/My-TTGO-Watch/files/7861434/My-TTGO-Watch.verbose.build.log)

I will try with other settings like moving the build output and renaming things.

closed 2022-01-27 by sharandac

clintonefisher, 2022-10-06:
FYI, this was broken for me despite removing apps from main.cpp and shortening paths.
The SOLUTION was to ALSO delete the app files from the /src/apps tree.

The README.md could be updated to reflect this, and probably save a lot of people some time and effort.